TheThingsArchive/ttn #265: Support adaptive data rate (ADR)

johanstokking, 2016-09-20:
Requires Network Server to register whether the node is mobile or static. ADR applies to static nodes only.

tftelkamp, 2016-09-22:
Some initial notes.

ADR is only applicable to static nodes that have stable radio performance to the gateway(s). Mobile devices should not use ADR, and define their own strategy (a.k.a. ‘blind ADR’). Note though that a node can be static at one time, and mobile at another. 

The nodes decide whether to use ADR or not (and when), not the network. A node that wants to receive ADR MAC commands will set the ADR bit in its uplink messages. If the network determines that the link is stable enough, it can adjust the nodes data rate, transmit power, and number of repetitions, using the (to be defined) ADR algorithm.

This algorithm will be applied to every node after every uplink, regardless whether the node wants ADR or not (but of course nothing is sent to nodes with ADR disabled). If the link is stable, and the network does have an ADR solution available, it wil set the ADR bit in downlink messages. If the link becomes unstable (too much variation in SNR for example), it will unset the ADR bit. The nodes can include this information in their strategy, but don’t have to use it. It is informational.

ADR updates, from the network, are to be expected on a daily or weekly basis, in a static and stable environment. Also not that ADR requires a downlink for every so many uplinks (configurable).

tftelkamp, 2016-09-22:
ADRACKReq: the network needs to send 'a' downlink within the next ADR_ACK_DELAY uplinks. Setting the ACK bit is not necessary (only needed in case of a confirmed uplink). This downlink message can be anything, initiated by the application, an ACK, MAC commands, or just empty payload frame.

LoRaWAN V1.0:

If an end-device whose data rate is optimized by the network to use a data rate higher than its default data rate, it periodically needs to validate that the network still receives the uplink frames. Each time the uplink frame counter is incremented (for each new uplink, repeated transmissions do not increase the counter), the device increments an ADR_ACK_CNT counter. After ADR_ACK_LIMIT uplinks (ADR_ACK_CNT >= ADR_ACK_LIMIT) without any downlink response, it sets the ADR acknowledgment request bit (ADRACKReq). The network is required to respond with a downlink frame within the time set by the ADR_ACK_DELAY, any received downlink frame following an uplink frame resets the ADR_ACK_CNT counter. The downlink ACK bit does not need to be set as any response during the receive slot of the end-device indicates that the gateway has still received the uplinks from this device.

tftelkamp, 2016-10-20:
** ADR **

This algorithm is based on the preliminary ‘_simple rate adaptation recommended algorithm_’ V1.0 by Semtech. Once published this document will be attached to this issue.

Applicable to EU868 only. Other regions will follow.

Only static nodes should do ADR. This can also be a mobile node that is ‘parked’ on a fixed spot, and able to detect this. The node decides if ADR will be used or not, not the application or network server.

The node that wants its data rate managed by the network, sets the ADR bit to 1 in its uplink messages.

On detecting this, the network server will start collecting the maximum SNR for each received message (so per frame counter), as well as the number of gateways (for future use), and keep track of the last 20 received uplinks. If the ADR uplink bit is 0, the table will be deleted, and build again when ADR becomes 1.

After 20 frames it can compute the SNR margin, using the data rate (DR) of the last received message:

The Max SNR over these 20 frames (SNRm) is used.

SNRmargin = SNRm – RequiredSNR(DR) - margin_db

Where Required SNR is:

DR5/SF7: -7.5
DR4/SF8: -10
DR3/SF9: -12.5
DR2/SF10: -15
DR1/SF11: -17.5
DR0/SF12: -20

and margin_db = 10dB (default, should be configurable, maybe even per node). 

Then the number of steps is computed as follows:

NStep = round(SNRmargin/3) 

If NStep > 0 the data rate can be increased and/or power reduced. If Nstep < 0, power can be increased (to the max.).

For NStep > 0, first the data rate is increased (by Nstep) until DR5 is reached. If the number of steps < Nstep, the remainder is used to decrease the TXpower by 3dBm per step, until TXmin is reached. TXmin = 2 dBm for EU868.

For Nstep < 0, TX power is increased by 3dBm per step, until TXmax is reached (=14 dBm for EU868).

Note: the data rate is never decreased, this is done automatically by the node if ADRACKReq’s get unacknowledged.

After sending the first LinkADRReq, the network will send an ADR request again if needed:

1. at most every Nth uplink frame (N to be decided)
2. only after ADR_ACK_Req is set by the device in uplink. The device will unset ADR_ACK_Req after receiving a ADRACKReq in addition to sending an LinkADRAns, hence providing two mechanisms to let the network know the ADR request has been received. If neither is done, the ADRACKReq might have been lost.

** Repetition rate **

The NbRep parameter is set based on the frame loss in the last 20 received frames.

The new NbRep is a function of frame loss and current NbRep value:

			Current value
			1	2	3

< 5%		1	1	2
5%-10%		1	2	3
10%-30%	2	3	3
> 30%		3	3	3

** Channel mask **

Main use for TTN will be to free up the 3 join channels, by selectively disabling these channels for uplink on certain devices after a join. (Note: ChMask is LSB?)

closed 2017-02-07 by htdvisser

amrbekhit, 2017-09-14:
Hi,

The algorithm mentions that after using NStep to increase the data rate, any remaining NStep is used to decrease the TX power. Alternatively, if NStep < 0, then TX Power is increased. This is then used to populate the LinkADRReq MAC message.

However, how do we know the current TX Power? As far as I can see, the end device has no standard way of transmitting what the current TX power is.

htdvisser, 2017-09-14:
The assumption is that a device starts transmitting at maximum power. After the first time the TX power is adjusted through ADR, the backend knows the exact settings that are used.

amrbekhit, 2017-09-14:
Ah I see thanks - so presumably you need to record the current TX power against each device and then refer to that in subsequent ADR calculations when it exists, right?

htdvisser, 2017-09-14:
Exactly. We store the TxPower index of a device after we do ADR and use that value for the calculations. The first time it's simply the default value of an `uint`, which is `0`.

oyj0594, 2017-09-21:
Hi @htdvisser 

Based on current simple ADR algorithm, do each node have to use confirmed message which requires ACK from server all the time?

However, as you know, downlink traffic is limited.
so, how can a LoRaWAN system deploy this mechanism?

johanstokking, 2017-09-21:
@oyj0594 confirmed uplink is not required. By default, end devices receive messages in RX1 and RX2 regardless of setting the ACK bit, to receive application downlink and/or MAC commands. Every `n` messages, according to the algorithm, the network server will send a MAC command for ADR, if ADR is requested.

PR #430 (2017-02-07): Implement Adaptive Data Rate
Resolves #265 

@johanstokking: please review

@tftelkamp: please review changes in the following files:

- `core/band/adr.go`
- `core/networkserver/adr.go`
- `core/networkserver/uplink_mac.go`

Remaining questions for @tftelkamp:

- Should we set the ACK bit on downlink messages?
- What can we do when the request is not accepted (one of the three ACKs in the LinkADRAns is 0)?
commit f0e2655e (2017-02-07): Merge branch 'feature/adr' into develop

Resolves #265