TheThingsArchive/ttn #270: Support private Handler

johanstokking, 2016-09-23:


romeovs, 2016-09-23:
Currently, the account server is oblivious to handlers, so I do not see issues with using a different handler. Things to consider:
- The namespace for app, gateway and component IDs will be shared with this handler as well, which might be counter-intuitive: "If it's my private handler, then why is the `test` app ID already taken?" It makes sense for us, but might not be what some people want/need.

Using the handler in dashboard is trickier currently: it uses the public discovery server to find public handler announcements and only let's a user pick from them.
We could build a field that also let's people use a private handler (eg. it's a dropdown/select, but people can add their own value `https://private.handler.com`). @htdvisser do you see other issue with using the private handler here (with certificates for example)?

htdvisser, 2016-09-23:
* App namespacing might be a good idea.
* External handlers must publish themselves to the discovery server in order to interact with the rest of the community network. We can add a flag to indicate whether each handler should be displayed in the dashboard or not.

htdvisser, 2016-10-25:
3801b6547c80151e6981ba527476e77111ecb862 adds a `public` bool to each discovery announcement. This can be used in the console to ignore non-public Handlers

closed 2017-01-11 by htdvisser