TheThingsArchive/ttn #272: Handle retries for confirmed uplinks

htdvisser, 2016-09-26:
[image]

matthijskooijman, 2016-11-16:
```
16:35 <htdvisser> There is one issue with confirmed uplink though, but in my opinion that's more a flaw in LoRaWAN itself. When a device does not receive an ACK to a confirmed uplink, it sends the message again (which is okay) with the same frame counter (which I think is wrong). The backend flags this as a possible replay attack, and will therefore ignore the message
18:17 <matthijs> htdvisser: Shouldn't it just drop the message *and* send an ACK?
18:45 <htdvisser> @matthijs: please comment on Github Issue so that this doesn't get lost in Slack historyhttps://github.com/TheThingsNetwork/ttn/issues/272 
```

avbentem, 2017-01-21:
> When a device does not receive an ACK to a confirmed uplink, it sends the message again (which is okay) with the same frame counter (which I think is wrong).

I guess an increased uplink frame counter would be troublesome as then backend and application would not even know they got a retry.

Just FYI, when fixing this: note that the image shows that the backend should use an increased downlink counter `{Cd+1}` when confirming the retry.

matthijskooijman, 2017-01-21:
Indeed, if the ACK went missing but the original upstream packet was received ok, the application will receive the same data twice. This is not a problem for timestamped sensor data, but for a data stream or otherwise non-idempotent data, this would force the application to add their own sequencing / frame counters into the data, which seems counter-productive.

commit b00b9876 (2017-02-01): Handle retries for confirmed uplinks

Resolves #272
PR #443 (2017-02-01): Handle retries for confirmed uplinks
Resolves #272 

Remaining question: What should the Handler do with retries?

- Just publish on MQTT (assuming the application checks the `counter` value)
- Publish on MQTT, but indicate that it's a retry (assuming the application will check for this flag)
- Don't publish on MQTT at all

Requires change in #442: now that we know a downlink (ACK) was not received by the node, we should also re-send the downlink message.
closed 2017-02-01 by htdvisser