TheThingsArchive/ttn #283: Improve TTNCTL api reference

FokkeZB, 2016-09-30:
@htdvisser I'd like to make some changes to the [generated](https://github.com/TheThingsNetwork/ttn/blob/refactor/ttnctl/cmd/docs/README.md) TTNCTL API reference.

- Skip the [Options inherited from parent commands](https://github.com/TheThingsNetwork/ttn/blob/refactor/ttnctl/cmd/docs/README.md#options-inherited-from-parent-commands) sections since they are the same everywhere.
- Start the API reference with the above options, since they originate from the `ttnctl` command.
- Remove the **Synopsis** heading and replace the line above it with the synopsis.
- Make the **Examples**, **Options** and other headings under the commands just **bold**, not actual headings.
- Make headings for subcommands like [ttnctl applications add](https://github.com/TheThingsNetwork/ttn/blob/refactor/ttnctl/cmd/docs/README.md#ttnctl-applications-add) one level deeper the the parent.

Example: https://gist.githubusercontent.com/FokkeZB/988d9673b1fcf3edc5322dc113aa887d/raw/580270417714f5fda9cbf78886c4fb5a4d130d84/example.md

johanstokking, 2016-09-30:
We use Cobra for ttnctl and this is the default behavior. Resolving this issue is not a matter of changing, but it's a matter of building our own style. For ttnctl, I think it's just fine. I'd rather leave ttnctl as compact as possible, also in terms of output.

FokkeZB, 2016-10-02:
Compactness is the very reason I propose these changes. The global options are repeated for every command and with all of them on the same heading level the TOC is waaaay to long: https://www.thethingsnetwork.org/docs/cli/#api-reference

It would just be a few regex replacements at https://github.com/TheThingsNetwork/ttn/blob/refactor/ttnctl/cmd/docs/generate.go#L36

htdvisser, 2016-10-03:
This won't be a problem. Will fix this :)

htdvisser, 2016-10-03:
Done in `refactor`. For `staging` you can use [README.md](https://github.com/TheThingsNetwork/ttn/files/506589/README.md.txt)

commit 641b6097 (2016-10-03): Update cli docs

Resolves #283

[Skip CI]
closed 2016-10-03 by htdvisser