TheThingsArchive/ttn #286: Enforce single-handler-per-app constraint

romeovs, 2016-10-05:
An app should only be able to register to one handler at the same time.

To enforce this, we could do several things:
- Have the account server store which handler the app is using. This is not preferred because the account server does not know about handlers yet, and introducing it to the discovery server is adding a lot of complexity.
- Enforce it whilst registering to a new handler. This is only feasible because the console is the central point that can enforce this, `ttnctl` would not be able to this however because it does not know which handlers the app was previously registered to.
- This could become part of the discovery mechanism. The discovery server could store which app is registered to which handler. A drawback is that the discovery server does not know about private handlers.

htdvisser, 2016-10-05:
The discovery server already keeps track of this, but as we don't have control over handlers, we can't be sure that they will delete an app when it is registered on another handler.

htdvisser, 2016-10-25:
Nothing more to do here. The discovery server keeps track of AppID-Handler mappings. It's the user's responsibility to delete devices from an old handler if he switches between handlers.

closed 2016-10-25 by htdvisser