TheThingsArchive/ttn #293: ttn/mqtt/topics ParseDeviceTopic Regex does not match

Niondir, 2016-10-07:
I get the message `Received message on invalid uplink topic: 70B3D57ED000100F/devices/70B3D58FF0030834/up`

```
func ParseDeviceTopic(topic string) (*DeviceTopic, error) {
	pattern := regexp.MustCompile("^([0-9a-z](?:[_-]?[0-9a-z]){1,35}|\\+)/(devices)/([0-9a-z](?:[_-]?[0-9a-z]){1,35}|\\+)/(events|up|down)([0-9a-z/]+)?$")
	matches := pattern.FindStringSubmatch(topic)
	if len(matches) < 4 {
		return nil, fmt.Errorf("Invalid topic format")
	}
```

It's the regex that does not to match the Topic. 

The Regex is missing uppercase letters.

Fix a)
Change the regexp: `^([0-9a-zA-Z](?:[_-]?[0-9a-zA-Z]){1,35}|\+)/(devices)/([0-9a-zA-Z](?:[_-]?[0-9a-zA-Z]){1,35}|\+)/(events|up|down)([0-9a-z/]+)?$`

Fix b)
Match to Lower: `matches := pattern.FindStringSubmatch(strings.ToLower(topic))`

**I prefer Fix a)**

Niondir, 2016-10-07:
Just an additional note. Maybe parsing the topic is not that important since the handler also has the appID and DevID available in `req.DevID` and `req.AppID` and thus it's redundant to fetch it from the topic. So maybe just parse the topic as a parameter so one can parse it himself.

johanstokking, 2016-10-07:
The app ID and device ID are case insensitive and they will be handled lowercase only. Uppercase characters in the device ID and application ID are invalid. The regular expression is correct.

I'm curious to how this message got published. Which revision are you using?

Niondir, 2016-10-07:
I'm using "tcp://staging.thethingsnetwork.org:1883" with `client.SubscribeUplink(handleUplink)` and `handleUplink(client mqtt.Client, appEUI string, devEUI string, req mqtt.UplinkMessage)`

Niondir, 2016-10-07:
```
Tarion@Tarion-PC MINGW32 /c/dev/gopath/src/github.com/TheThingsNetwork/ttn (v2-preview)
$ git status
On branch v2-preview
Your branch is up-to-date with 'origin/v2-preview'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
```

johanstokking, 2016-10-07:
OK, you might be mixing `v1-staging` and `v2-preview`. We have changed this behavior. See the different branches.

closed 2016-10-07 by johanstokking

Niondir, 2016-10-07:
Mhm I just did `go get` to get the dependency. You mean I just have a newer version of the client than the current staging system? 

I will investigate into version pinning a bit and try to fix it by taking the right version. Thanks.

Niondir, 2016-10-07:
@johanstokking I switched to v1-staging - same issue.

With `tcp://eu.thethings.network:1883` I get `Not Authorized`

I will dig into it later.

johanstokking, 2016-10-07:
Use the `v1-staging` MQTT client, e.g. ttnctl, with `staging.thethingsnetwork.org`. Use the `v2-preview` MQTT client, e.g. ttnctl, with `eu.thethings.network`. The application and device registry are separate, so that explains the authorization issue. See also https://thethingsnetwork.org/docs.