TheThingsArchive/ttn #295: Device not found

johanstokking, 2016-10-10:
Although the device appears in the list, info cannot be shown. See the `list` and `info` command output.

The AppEUI may have changed with the same device name.

```
➜  ~ ttnctl devices list
  INFO Using Application                        AppEUI=70B3D57EF0000047 AppID=x
  INFO Discovering Handler...                   Handler=ttn-handler-eu
  INFO Connecting with Handler...               Handler=eu.thethings.network:1904

DevID  	AppEUI          	DevEUI          	DevAddr 	Up/Down
y	70B3D57EF0000047	0059AC00001503D9	26001FEC	0/0

  INFO Listed 2 devices                         AppID=x
➜  ~ ttnctl devices info y
  INFO Using Application                        AppEUI=70B3D57EF0000047 AppID=x
  INFO Discovering Handler...                   Handler=ttn-handler-eu
  INFO Connecting with Handler...               Handler=eu.thethings.network:1904
 FATAL Could not get existing device.           error=Could not get device from Handler: Broker did not return device: NetworkServer did not return device: 70B3D57EF0000047/0059AC00001503D9 not found
```

romeovs, 2016-10-10:
@johanstokking tested wether or not changing the app EUI has an influence on the device and, as far as I can see, this is not the issue.

htdvisser, 2016-10-10:
For some reason this device was not in the database of the networkserver. No idea how that is possible, because the handler should not save a device when it is rejected by the networkserver.

Unfortunately we don't have server logs showing the history of this device. Without knowing what happened to the device entry in the networkserver database, I am not able to reproduce this issue.

johanstokking, 2016-10-10:
Is it not there at all, or is it there with a different AppEUI or DevEUI? I.e., are there records in the Handler with other EUIs which do exist on the Network Server?

What about deleting; is it possible that the device is deleted successfully from the Network Server, but not in the Handler?

htdvisser, 2016-10-11:
This issue was caused by the following:

- Create a device "A" with AppEUI `70B3D57EF0000047` and DevEUI `0059AC00001503D9`
- Create a device "B" with AppEUI `70B3D57EF0000047` and DevEUI `0059AC00001503D9`
- Delete device "A"
- Get device "B"

Obviously, it should not be possible to register two devices with the same AppEUI/DevEUI. This was fixed in 217e2a8b4b71adc3039755b3b987ec8803c760cb and will be deployed to the servers later today.

closed 2016-10-11 by htdvisser