TheThingsArchive/ttn #299: Payload Functions: Pass bytes as Buffer to Decoder, not Array

FokkeZB, 2016-10-10:
The Decoder currently gets an Array (tested via `Array.isArray(bytes)` vs `Buffer.isBuffer(bytes)`). It would be very helpful if we'd pass the actual Buffer. You can still use it as an Array, but also get access to all the [read* methods](https://nodejs.org/api/buffer.html#buffer_buf_readdoublebe_offset_noassert) to get multiple bytes as  numeric value without having to do manual bit shifting.

FokkeZB, 2016-10-10:
BTW, I cannot create a buffer in the Payload Function either. If I run `Buffer.isBuffer(bytes)` I get:

```
ReferenceError("'Buffer' is not defined")
```

FokkeZB, 2016-10-10:
This is how it would simplify decoders:

Before:

```js
var myInt32 = (bytes[0] << 24) + (bytes[1] << 16) + (bytes[2] << 8) + (bytes[3]);
var myInt16 = (bytes[4] << 8) + (bytes[5]); 
// var myFloat =  IMPOSSIBLE?
```

After:

```js
var myInt32 = bytes.readInt32BE(0);
var myInt16 = bytes.readInt16BE(4);
var myFloat = bytes.readFloatBE(6);
```

johanstokking, 2016-10-10:
The payload functions are JavaScript without browser context, it's not Node.js so there's no `Buffer` available. I'd rather pre-load a JavaScript library that makes life a bit easier. Submit a separate issue for that with specific requirements or API.

closed 2016-10-10 by johanstokking

FokkeZB, 2016-10-11:
See https://github.com/TheThingsNetwork/ttn/issues/300

FokkeZB, 2016-10-11:
I also noticed otto is missing simple APIs like `Array.shift()` as this:

```js
function Decoder(bytes /*, port */) {
  var decoded = {};

  // until PF support port we pass it in first byte
  decoded.port = bytes.shift();
}
```

will give:

```
TypeError("TypeError")
```

Maybe instead of using a very limited JS VM, we should just use Go for the payload functions?

https://github.com/novalagung/golpal

johanstokking, 2016-10-11:
I think that JavaScript is the most commonly understood language. I'd rather work towards a Node.js container than Go, but best is to keep it simple. There are many a lot of advantages to have a lightweight vm like this when it comes to scaling. Also, the payload functions in general are remote code execution which is always risky.

Ideally, developers use Google Protocol Buffers so they don't have to deal with arrays at all. If they want to deal with arrays, they can 1) use our utility functions that we pre-load, 2) do the magic themselves in the payload functions or 3) do the magic themselves on their premises. For private handlers, we may support 4) a Go function and 5) a HTTP call to decode/convert/validate.

FokkeZB, 2016-10-11:
JS is indeed lowest barrier. But if it's missing even simple (browser) APIs like `Array.prototype.shift()` then it will be frustrating to work with. I'd love to have NodeJS if we can.

johanstokking, 2016-10-11:
Jep, let's try to get the max out of Otto. Node.js is a completely different setup.

Seems like there is support for `shift()`: https://github.com/robertkrimen/otto/blob/master/builtin_array.go#L92-L112

Don't know why it's not available as you expect it. Can you isolate this issue a bit more maybe?

FokkeZB, 2016-10-11:
It's a bug: #303