TheThingsArchive/ttn #300: Provide NodeJS Buffer like APIs to Decoder and Encoder

FokkeZB, 2016-10-11:
The Decoder currently gets an Array, not a Buffer because the [otto JS vm](https://github.com/robertkrimen/otto) we use does not know the NodeJS [Buffer](https://nodejs.org/api/buffer.html) API (see #299).

The NodeJS Buffer API has some really nice [read/write* methods](https://nodejs.org/api/buffer.html#buffer_buf_readdoublebe_offset_noassert) to read and write numbers of different formats from/to an offset in the buffer. This would remove the need for manual bit shifting, which puzzles many (new) users.

Before:

```js
var myInt32 = (bytes[0] << 24) + (bytes[1] << 16) + (bytes[2] << 8) + (bytes[3]);
var myInt16 = (bytes[4] << 8) + (bytes[5]); 
// var myFloat =  IMPOSSIBLE?
```

After:

```js
var myInt32 = bytes.readInt32BE(0);
var myInt16 = bytes.readInt16BE(4);
var myFloat = bytes.readFloatBE(6);
```

What we should look into is loading a browser JS implementation of Buffer:

https://github.com/feross/buffer

It is quite big, but provides a lot of convenience to users. We could also wright a lighter wrapper around the underlying [ieee754](https://github.com/feross/ieee754/blob/master/index.js) with just the read/write methods.

Ideally we'd pass `Buffer.from(bytes)` to the Decoder so the user can work with it right away.

In the Encoder we could pass an empty buffer as 3rd argument which the user can fill (no need to return) using the write methods.

johanstokking, 2016-10-11:
I prefer a lightweight implementation on top of `ieee754` for reading and writing signed/unsigned 8/16/32 bit integers little/big endian, booleans, and bit sets and reads.

FokkeZB, 2016-10-11:
Me too, and we should then have the same API on the Arduino. Provides a nice level in between protobufs and bytes.

johanstokking, 2016-10-11:
Yes. I would, however, not change the Encoder as it is for portability. Let it return a byte array, so simply `return buffer.toArray();`. People are not entirely stupid and it avoids a lot of magic in the Handler to add a custom parameter and check for return type.

htdvisser, 2017-01-31:
@AntoineRondelet, @romeovs?

johanstokking, 2017-03-02:
We can use Google Protocol Buffers or [Cayenne Low Power Payload](https://mydevices.com/cayenne/docs/#lora-cayenne-low-power-payload)

closed 2017-03-02 by johanstokking