TheThingsArchive/ttn #302: v2-preview ABP no longer working

kruisdraad, 2016-10-11:
Situation: I have a device 527CB036 which keys, eui's, addr's, etc are exactly the same on my own backend as well as on TTN. I have a node thats pinging a GW and its counter is now over 67294. 

The backend now reports an error:
device that validates MIC not found

When i throw the device away it gives another error:
Device with DevAddr 527CB036 and FCnt <= 1748 not found

Now there is the thing, that FCnt is 1748 and going up, while on TTN its 67294, looking at the device its also on that high number. Even though the flag --disable-fcnt-check is set i keep getting the MIC error.

also it seems there is no 'unset' option so you cant disable --32-bit-fcnt or --disable-fcnt-check flags?

kruisdraad, 2016-10-11:
resetting the device (fCnt 0) and it works again. looks like a bug with decoding the lora packet and having a high counter

htdvisser, 2016-10-11:
The `--32-bit-fcnt` and `--disable-fcnt-check` flags should not be used together. In case of high counters, the backend will assume overflow in the the 16 bit counter when it sees a counter around `0`, and will increase the frame counter.

- 27f09b994af6b8c19bf7edeaf80c60861f4ceffc adds the "unset" options for `ttnctl devices set`
- e83a1772a47a10bd505d10eac9cf0f7baa0003d8 adds a warning for combining `32BitFCnt` with `DisableFCntCheck`

These changes should fix the problem you're having. If so, please close the issue.

kruisdraad, 2016-10-11:
just to be clear, there is no --32-bit-fcnt set, i never touched that option

if you look at https://github.com/TheThingsNetwork/ttn/issues/301 you see no options were set at all except DisableFCntCheck

I am running the node again its gonna take a few days until it reaches 65k again

to be continuee

htdvisser, 2016-10-25:
In a Slack conversation, you wrote "jah, in v2 gaat het goed, :slightly_smiling_face:", so I guess this issue can be closed.

closed 2016-10-25 by htdvisser