TheThingsArchive/ttn #305: Payload Functions: Allow additional functions called from PF

FokkeZB, 2016-10-11:
We currently wrap the payload functions in `(%s)(payload)` when we execute them:
https://github.com/TheThingsNetwork/ttn/blob/v2-preview/core/handler/convert_fields.go#L70

This makes that if you add another function after your function to do something like:

```js
function Decoder(bytes) {
  return {
    foo: bar(bytes)
  };
}

function foo() {
  return 'alpha';
}
```

You get:

```
Error("(anonymous): Line 7:1 Unexpected token function (and 5 more errors)")
```

We could instead wrap the code in:

```js
%s;
Decoder(payload);
```

Or go common-JS-ish with:

```js
exports = {};
%s;
exports.Decoder(payload);
```

johanstokking, 2016-10-11:
Or declare the function in the function

FokkeZB, 2016-10-11:
That's an option, but not best practice and I rather see people not run into error and then having to guess that's a workaround

johanstokking, 2016-10-11:
Yes, we can go for the first approach you propose, where we have to teach the users to use `Decoder` as function name, but that's definitely doable.

FokkeZB, 2016-10-12:
Agree, then it's not a breaking change.

romeovs, 2016-10-25:
What about this?
```
Decoder = %s
Decoder(payload, port)
```

FokkeZB, 2016-10-25:
Even better!

romeovs, 2016-10-25:
On the other hand, that will prohibit things like:
```js
var foo = 10;
function Decoder (payload) { /* ... */ }
```

FokkeZB, 2016-10-25:
o shoot, true...

PR #332 (2016-10-27): Allow for additional code by calling named payload function
Closes #305 by using @FokkeZB 's proposal.
closed 2016-10-27 by htdvisser