TheThingsArchive/ttn #310: Simulate uplink in Handler

htdvisser, 2016-10-14:
Instead of `ttnctl` simulating a message through the entire network, it should be able to send a simulated uplink message directly to the Handler, which will then send it over MQTT.

This will help application developers test their applications without having to worry about rate limiting of simulated devices.

Niondir, 2016-11-09:
Is there a way to send a uplink message via `ttnctl` right now? Or is it planned? I'm working on something like that to better understand the protocol and TTN infrastructure. But I like to send the message through the entire network.

htdvisser, 2016-11-10:
Yes, there's the `ttnctl uplink` command. Add the `--help` flag to learn more.

Niondir, 2016-11-10:
Funny it's there but not listed:

```
$ ttnctl help
ttnctl controls The Things Network from the command line.

Usage:
  ttnctl [command]

Available Commands:
  applications Manage applications
  config       Print the used configuration
  devices      Manage devices
  downlink     Send a downlink message to a device
  gateways     Manage gateways
  selfupdate   Update ttnctl to the latest version
  subscribe    Subscribe to events for this application
  user         Show the current user
  version      Get build and version information

```

But here it shows up :)
```
$ ttnctl uplink
Usage:
  ttnctl uplink [DevAddr] [NwkSKey] [AppSKey] [FCnt] [Payload] [flags]

Flags:
      --confirmed           Use confirmed uplink (this also sets --downlink)
      --downlink            Also start downlink (unstable)
      --gateway-id string   The ID of the gateway that you are faking (you can only fake gateways that you own)

Global Flags:
      --auth-server string         The address of the OAuth 2.0 server (default "https://account.thethingsnetwork.org")
      --config string              config file (default is $HOME/.ttnctl.yml)
      --data string                directory where ttnctl stores data (default is $HOME/.ttnctl)
  -d, --debug                      Enable debug mode
      --discovery-address string   The address of the Discovery server (default "discover.thethingsnetwork.org:1900")
      --handler-id string          The ID of the TTN Handler as announced in the Discovery server (default "ttn-handler-eu")
      --mqtt-address string        The address of the MQTT broker (default "eu.thethings.network:1883")
      --mqtt-password string       The password for the MQTT broker
      --mqtt-username string       The username for the MQTT broker
      --router-id string           The ID of the TTN Router as announced in the Discovery server (default "ttn-router-eu")
```

Niondir, 2016-11-10:
It's `Hidden: true,`

htdvisser, 2016-11-10:
Yes, this should only be used for backend development. We don't want everyone to use this to debug their nodes/applications ;)

commit a6379fca (2017-02-01): Implement uplink simulation in Handler

Resolves #310
PR #444 (2017-02-01): Simulate uplink messages directly in Handler
Resolves #310
closed 2017-02-01 by htdvisser