TheThingsArchive/ttn #314: Dedicate port for nodes to ping for diagnostics

FokkeZB, 2016-10-18:
On a computer I can see I'm connected to Wifi...
On a smartphone I can see I have coverage...

On a node.. I get successful transmission and still don't know

What if we'd:

1. Dedicate a port (255? 100?) that returns diagnostics
2. Provide a `ttn.ping()` API that sends a byte to this port and decoded the diagnostics but also has client-side logic to help diagnose failures that cause the message not to make the backend
3. Document the format of the diagnostics for nodes we don't have a lib for

The diagnostics could simply be different success/error codes like:

1. All good
2. Cannot reach any gateway
3. Wrong keys
4. ... ?

/cc @johanstokking

htdvisser, 2016-10-18:
Unfortunately it's not possible to tell a node (from the network side) what exactly went wrong. LoRaWAN was just not built for this.

situation 1: Messages just show up in the dashboard
situation 2: LoRaWAN is a radio protocol, so if it cannot reach any gateway, there is no response.
situation 3: If wrong keys are used, we don't know how to encrypt a response and sign the MIC, so even if we would send something back, the node won't be able to read it.
network error: Apparently there's a communication issue, so we either didn't receive anything on the server (and we can't reply) or we sent a reply, but it doesn't arrive back at the node
any other situation: We might be able to publish errors to MQTT, but that's it.

For checking connectivity, there's the LinkCheck MAC command (which is supported by the v2 backend). This works, but is probably not the best way to spend your downlink quota.

johanstokking, 2016-10-18:
Apart from confirmed uplink and link check MAC command, we can't do very much today, especially not when it comes to reporting status to the node.

In the TTN model, the the `NwkSKey` is known by the Network Server and the `AppSKey` is known by the Handler, so if there goes anything wrong in the Router, Broker, Network or Discovery Server, there's no way the Router can respond.

Instead, we'd better work on the NOC and get better insight in the data flow. Ideally, we'll be able to follow each message from component to component, or when and why a component drops it. The next step would be to make this available to the user in the Console.

FokkeZB, 2016-10-18:
Haha, I was afraid that would be the answer, so really the thing we need is:

> Provide troubleshooting for every possible point of failure

Let's identify every point of failure and think of manual and technical ways to help the user determine what the point of failure is and how to fix that.

Probably more a docs issue then, so let's build a list of point of failures, ordered from first to last one to check with solutions here:

https://github.com/TheThingsNetwork/docs/wiki/Device-Troubleshooting

This will also clarify where the NOC and surfacing its info in the Console have a role.

johanstokking, 2016-10-18:
Here's a start:

### ABP
1. Wrong `DevAddr` or wrong `NwkSKey`: device unknown or MIC check fails, Broker drops message. Possible causes: wrong key, confused MSB/LSB. Solution: configure correctly
2. Wrong `AppSKey`: Handler produces garbage. Possible causes: wrong key, confused MSB/LSB. Solution: configure correctly
3. Colliding `DevAddr`, `NwkSKey` or `AppSKey`: you see data from other devices as well. Solution: configure unique set of address and keys (e.g. use our randomizer)
4. Frame counters reset: no more frames, except when frame counters comes above previous value, Broker drops the message. Possible causes: power reset and no relax frame counters on. Solution: switch to OTAA if power reset always causes frame counter reset, or disable frame counter checks (unsafe)

### OTAA
1. Wrong `AppEUI`: the Handler cannot be reached for activation, all Brokers drop. Possible causes: wrong EUI, confused MSB/LSB, Handler is not registered on Broker. Solution: configure correctly, make sure Handler is registered (shouldn't be a problem in the public community network)
2. Wrong `AppKey`: the challenge by the Handler of the Broker(s) failed, the Broker(s) will drop the message. Possible causes: wrong key, confused MSB/LSB. Solution: configure correctly
3. Wrong `DevEUI`: unknown device in NS, Broker drops the message. Possible causes: wrong EUI, confused MSB/LSB, device not registered by the Handler. Solution: configure correctly, register device
4. Duplicate `AppEUI`, `DevEUI` and `AppKey`: device gets activated in two networks, it's random which join accept the device processes. Possible causes: registering the same device on two networks. Solution: don't do that, use LoRaWAN 1.1 join servers (not fully specified yet). Note: we're already avoiding this in the public community network
5. Activation still doesn't work: the activation event might be fired by the Handler, but the join accept may not be received by the device, see 1 in *General* below

### General
1. Confirmed uplink doesn't acknowledge: there's no coverage, there's no gateway with downlink time left (or no downlink gateway at all, e.g. single channel gateway), gateway transmission collision or gateway network issue
2. (Un)confirmed data is not in Console: there's no coverage, gateway network issue

FokkeZB, 2016-10-18:
Thanks! I'll rewrite it to go from experienced issue to cause. It'll be important to have extensive troubleshooting in place before we ship to not get overwhelmed with support requests.

closed 2018-01-17 by htdvisser