TheThingsArchive/ttn #316: Payload functions: Validate code

kruisdraad, 2016-10-18:
When creating a payload function adding bogus code will just not run, and then outputting the default. However you have no idea the code is bad, If a function does not work the lines should perhaps turn red with a message the function failed (perhaps a reason?)

johanstokking, 2016-10-18:
Are you referring to `v1-staging` or `v2-preview`? In the latter, the default decoder is:

```js
function Decoder(bytes) {
  // Decode an uplink message from a buffer
  // (array) of bytes to an object of fields.
  var decoded = {};

  // decoded.led = bytes[0];

  return decoded;
}
```

Which runs, and really there's no default output possible here. You'll get the `payload_raw` anyway on MQTT, regardless of the payload functions.

We have an issue open that requires the user to test the payload function before saving.

htdvisser, 2016-10-19:
> We have an issue open that requires the user to test the payload function before saving.

We should probably also add this in `ttnctl`.

closed 2017-02-01 by htdvisser