TheThingsArchive/ttn #317: No way to tell to which device an ack belongs

romeovs, 2016-10-19:
This is a **feature request** for **MQTT** (**v2-preview**).

- Explain what you want to do
I want to listen to all acknowledgements for all devices of a specific app, and know to which device each acknowledgement belongs.

- Explain what steps you took
Listened to `<AppID>/devices/#/events/ack`

- Explain what went wrong or what is missing
I get the acks as `{}`, so I do not know which device they belong to.

romeovs, 2016-10-19:
Just updated the title, as the same thing goes for 
- `<AppID>/devices/#/events/up/errors`
- `<AppID>/devices/#/events/down/errors`
- `<AppID>/devices/#/events/activations/errors`

htdvisser, 2016-10-19:
You can get the device ID from the MQTT topic.

closed 2016-10-19 by htdvisser