TheThingsArchive/ttn #341: Sometimes it takes hours until node packages are being received

telmich, 2016-11-03:
This is a **bug report** **the Backend** (**v1-staging**).

- Explain what you want to do

Send data from a node

- Explain what steps you took

Flashed the code ( https://github.com/telmich/lorawan/tree/master/sodaq_one)
to Sodaq One, started device.

- Explain what went wrong or what is missing

For hours I run into the case NotConnected (see line 79 in https://github.com/telmich/lorawan/blob/master/sodaq_one/lora.ino) and then it suddenly "works".

I connect using OTAA with the correct deveui, appkey & co.

This is not 100% reproducible, but has happened at least 5 times for me. Leaving the device on during night at some point fixes the problem.

This happens with 3 different Sodaq nodes. Connecting to the swisscom network instantly works

## Environment

- My application:
  - App ID: `hackzurichtest`
- My gateway:
  - EUI: `B827EBFFFEC267C7`
- My node:
  - Device ID: ``
  - AppEUI: `70B3D57ED0000E9B`
  - DevEUI: `9CD90BB52B6A1D02`
- I am using the `Sodaq` library for the `Arduino/C` programming language

htdvisser, 2016-11-03:
When did this problem occur exactly? We had some downtime yesterday evening (CET) that could explain this issue.

telmich, 2016-11-03:
The last time was actually yesterday (sic!), I will try to report it again when I see it. I was suspecting the frame counters to be a problem, because they *might* be reset by sodaq one on restart.

htdvisser, 2016-11-03:
With OTAA devices, frame counters are reset when the device joins. This is conform the specification and will therefore not lead to any issues. Please reopen this issue or create a new one if you see this problem again.

htdvisser, 2016-11-03:
FYI: you can join the #ops channel on Slack to report (and see reports of) outages.

closed 2016-11-03 by htdvisser