TheThingsArchive/ttn #343: Make "Join not accepted: denied" more specific

FokkeZB, 2016-11-04:
Related to #314

There are multiple reasons why you could get "Join not accepted: denied".

We need to make this error more specific:

- Was there no gateway within reach?
- Were the keys incorrect?
- Was the downlink with session keys not received in response?

I know some of this is impossible with radio technology, but I predict this will be the top FAQ we'll get after we ship. So anything we can do on either the error returned to (and logged by) the Node or in the Console would be welcome.

johanstokking, 2016-11-04:
Report in the Arduino device library

FokkeZB, 2016-11-05:
Done: https://github.com/TheThingsNetwork/arduino-device-lib/issues/131

closed 2016-11-05 by FokkeZB