TheThingsArchive/ttn #353: Add ExchangeDeclarePassive to AMQP in Handler

htdvisser, 2016-11-09:
The Handler currently panicks when started with Docker

```
FATAL Could not initialize handler             error=Failed to bind queue ttn-handler-downlink with key *.devices.*.down on exchange 'ttn.handler' (Exception (404) Reason: "NOT_FOUND - no exchange 'ttn.handler' in vhost '/'")
```

This can be easily avoided by calling `ExchangeDeclarePassive` to declare the exchange if it doesn't exist yet.

johanstokking, 2016-11-10:
Yes, note that this needs the Handler to have permissions to declare exchanges. Might be fine for now, but we might choose for an elevated initialization process that sets up and migrates the environment, so that the actual components that have external interfaces (like the Handler), only have the rights that they actually need. Should the Handler, or other components with many permissions, get compromised, the impact is limited because of limited permissions.

htdvisser, 2016-11-10:
Isn't this exactly what ExchangeDeclarePassive is for? From the docs: `it sets the "passive" attribute to true. A passive exchange is assumed by RabbitMQ to already exist, and attempting to connect to a non-existent exchange will cause RabbitMQ to throw an exception. This function can be used to detect the existence of an exchange.`

johanstokking, 2016-11-10:
Yes that is what it's for, but my point is that the Handler should not have privileges to declare exchanges in the first place. It's not only using `ExchangeDeclarePassive`, which is fine in itself, but it should be an indicator to the user to declare the exchange out of band, or deliberately change the Handler's configuration to use a privileged account.

htdvisser, 2016-11-10:
I believe it's important to help contributors get started as smoothly as possible. This is also the reason why we have development keys and certificates in the repository. Setting up accounts with correct privileges is something we should document for the setup of production environments, but for development, I really believe that people should be able to get started with one keypress.

commit 8a7ee640 (2017-02-01): Assert presence of AMQP exchange when Handler starts

Resolves #353
closed 2017-02-01 by htdvisser