TheThingsArchive/ttn #355: Docs: HTTP API Reference clarifications

FokkeZB, 2016-11-10:
There's a few things we should clarify in https://github.com/TheThingsNetwork/ttn/blob/v2-preview/api/handler/HTTP-API.md, which is published at https://www.thethingsnetwork.org/docs/draft/platforms/#http-api-reference:

In general:

- What is the end point of this API?
- What error codes can I expect?
- Shouldn't we use PUT to update an existing object?

> We recommend everyone to use the gRPC interface if possible.

- How? Where can I find docs?

> This header should either contain a Bearer token with the JSON Web Token issued by the account server or a Key that is issued by the account server and can be exchanged to a JSON Web Token.

- How do I get that JSON Web Token? Where do I get that Key? How do I exchange that Key for a JSON Web Token?

/cc @htdvisser

htdvisser, 2016-11-17:
- The endpoint of the API depends on the deployment. We can document this for the Public Community Network in TheThingsNetwork/docs.
- Regular HTTP response codes
- Added PUT endpoints in `6afcdfa`
- Already documented in TheThingsNetwork/docs
- Already documented in TheThingsNetwork/docs

closed 2016-11-17 by htdvisser

FokkeZB, 2016-11-17:
Thanks, could you let me know the public community network endpoints to put in the docs?

FokkeZB, 2016-12-13:
@htdvisser Do you have those endpoints for the public community network for me to include in the docs?

htdvisser, 2016-12-13:
The components hosted by The Things Network Foundation use the default ports.
Discovery gRPC API has default port 1900 - HTTP proxy has default port 8080 - hostname `discover.thethingsnetwork.org`
Handler gRPC API has default port 1904 - HTTP proxy has default port 8084 - hostname can be found in Discovery (for The Things Network Foundation this is the same hostname as MQTT)