TheThingsArchive/ttn #356: Docs: Document Handler AMQP API

FokkeZB, 2016-11-13:
We need documentation (usage) and an API reference for [AMQP](https://github.com/TheThingsNetwork/ttn/tree/v2-preview/amqp) like we have for [mqtt](https://github.com/TheThingsNetwork/ttn/blob/v2-preview/mqtt/README.md), published on https://www.thethingsnetwork.org/docs/current/mqtt/.

htdvisser, 2016-11-14:
Will come after https://github.com/TheThingsNetwork/ttn/issues/325

krital, 2016-11-23:
If the feature is not urgent i might be able to assist, would be a good opportunity to get familiar with the API...

Alex

htdvisser, 2016-11-23:
Thanks for the offer, @krital! We first want to fully implement the AMQP API (see #325) before we'll document it.

krital, 2016-11-23:
I understand, my offer stands for both ;-)

htdvisser, 2016-11-23:
Oh your help is very welcome! I'll put both issues on "in progress" and will try to assign them to you.

closed 2017-05-08 by Sypheos