TheThingsArchive/ttn #363: Make ttnctl (and other) OAuth requests RFC compliant

kruisdraad, 2016-11-17:
The clientid TTNCTL uses no password(secret) to exchange tokens. However this requires 'public mode' to be enabled. RFC http://tools.ietf.org/html/rfc6749#section-4.4 ... so use secret: ttnctl which is just as easy and don't seem to confuse some proxy/cNAT servers.

romeovs, 2016-11-17:
@kruisdraad What do you mean with "public mode"? I cannot seem to find any mention of it in the RFC. 

To my understanding the RFC actually specifies that for public or native clients (like `ttnctl`) the Implicit Flow  should be used. (Or the [PKCE](https://tools.ietf.org/html/rfc7636) extension). 

Do you mean you want ttnctl to use the Client Credentials Flow (username + password)? We moved away from that because we want to implement 2FA in the future.

 [PKCE](https://tools.ietf.org/html/rfc7636) might be nice, but that's also something for the future.

kruisdraad, 2016-11-17:
problem list in that the pasword (secret) in the client cli should be used

in addition i found another RFC issue:

 The word "bearer" now has a capital "B" in JSON output to match OAuth 2.0 spec

htdvisser, 2016-11-30:
`ttnctl` now uses the (not-so-secret) "secret" `ttnctl`.

commit e6d97ebd (2016-11-30): Use oauth secret for ttnctl

Closes #363
closed 2016-11-30 by htdvisser