TheThingsArchive/ttn #370: Global questions about distributed network

cambierr, 2016-11-24:
Hey guys,

I know the backend v2 is still in dev, but I'd like to centralize some informations here on how to use/integrate it, even in a very early-POC:

1. When the handler announce itself to the discovery, should it already give the APP_IDs it's linked to or can it register them later ?
2. If an handler re-announce itself without the metadata, will the existing metadata be removed ?
3. When the handler subscribes to a broker, what's the best way to choose the broker to use ? Should we use the lowest latency ? or is there any information about the load of a broker ? or finally, are they load-balanced meaning that we should use the one of our geographic zone without caring about load balancing & HA ?
4. From the way metadata or query-able in discovery, won't we encounter huge performance drops when sending a get request returning all handlers + all their respective metadata, for instance ? I imagine the case where the EU handler will have millions of APP_IDs registered: Shouldn't we have a method like GetAllByMetadataFilter (I'm sure you understand what I mean here).

Thank you guys for your feedback on this !

htdvisser, 2016-11-24:
I'll give some short answers to these questions here, we will write more detailed documentation when the backend is actually ready for private handlers (#270).

1. A component has to announce itself once. The `Announce` method does not process Metadata (such as APP_IDs). Metadata should be added/removed with the `AddMetadata` and `DeleteMetadata` methods (I added a comment about this in the proto file).
2. As a result of this, a component can re-announce itself (for example, a changed IP address) without losing Metadata.
3. The Handler chooses a Broker based on region. So, for example, the public `ttn-handler-eu` connects to `ttn-broker-eu`. A Handler can only connect to *one* Broker, so if you want to connect to a private Broker as well as a public Broker, you'll need to run two separate Handlers. Load balancing is currently not implemented.
4. For performance reasons, we decided that components should keep a complete view of the network in memory (so a Router knows all Brokers, and a Broker knows all Handlers). This way, when a message for AppID `foo` should be handled, we can determine the Handler within a couple of microseconds instead of tens of milliseconds. Indeed, a GetAll request takes quite a lot of time (currently around 70ms), so we want to take a push-based approach instead (#369).

Let me know if you have any further questions. If this helps you enough, please close the issue.

cambierr, 2016-11-24:
Yeah, I now have everything in hands to build my poc. Thank you for your blazing-fast response!

closed 2016-11-24 by cambierr