TheThingsArchive/ttn #373: Changing "Frame counter checks" setting causes Device Address change

Paul-B, 2016-12-04:
This is a bug report for v2-preview.

- Explain what you want to do
Toggle the ""Frame counter checks" setting
- Explain what steps you took
Did that a couple of times
- Explain what went wrong or what is missing\
Found out that my Device Address has changed two a new value two times

## Environment

- `ttn version` returns: `Commit={...}`
- `ttnctl version` returns: `Commit={...}`
- My application:
  - App ID: `{sodaq_testing}`
- My gateway:
  - EUI: `{...}`
  - Connected to: `router.{...}.thethings.network:1700`
- My node:
  - Device ID: `{jpmtesting}`
  - AppEUI: `{70B3D57EF0000187}`
  - DevEUI: `{00E082F13BD06420}`
- MQTT is connected to: `{...}.thethings.network:{1883/8883}`

romeovs, 2016-12-05:
Fixed in the console! @Paul-B can you check this and close the issue if you agree?

Paul-B, 2016-12-05:
It still happens here. I don't know if you could confirm the bug but maybe I should double check if what I'm seeing is intended behavior.

romeovs, 2016-12-05:
Does this happen for OTAA devices or ABP devices?

Paul-B, 2016-12-05:
After trying again a couple of times I can confirm that the bug has been fixed. I must have still had the old version in my browser cache.

closed 2016-12-05 by Paul-B

romeovs, 2016-12-05:
Okay cool!

vikram-kadian, 2017-12-28:
Same happened, my device id changed when i enabled the frame counter checks from android app of ttn.

htdvisser, 2017-12-28:
We don't have an official Android app. Any Android or iOS apps are developed independently of TTN, so you'll have to contact the developer of those apps to get this solved.

vikram-kadian, 2017-12-28:
Oh! ok i see, have installed this app : https://play.google.com/store/apps/details?id=com.async.ttn&hl=en, and now have to reprogram my nodes to match the device address ABP :(

johanstokking, 2018-01-04:
@vikram-kadian yep, that's a community app, not developed by the core team.