TheThingsArchive/ttn #374: Inconsistencies in the account server API

cambierr, 2016-12-07:
This is a **improvement suggestion** for **account server** (**v2-preview**).

When getting an application, we have a response like this one:
`{
	"id": "test",
	"name": "test",
	"euis": [...],
	"created": "...",
	"rights": [...],
	"collaborators": [
		{
			"username": "cambierr",
			"email": "romain@shareif.com",
			"rights": [
				"settings",
				"delete",
				"collaborators",
				"devices"
			]
		}
	],
	"access_keys": [...]
}`

The interesting part is:
`"collaborators": [
		{
			"username": "cambierr",
			"email": "...",
			"rights": [
				"settings",
				"delete",
				"collaborators",
				"devices"
			]
		}
	]`

Where we can get an array containing Collaborator objects.

But when listing the actual collaborators of the application via get /applications/id/collaborators, we have a response like this one:
`{
	"cambierr": [
		"settings",
		"delete",
		"collaborators",
		"devices"
	]
}`

I think it would be great if we could use the same response format (the first) for both endpoints since it would make our (sdk-developers) work easier.

cambierr, 2016-12-07:
Btw, same goes for the PUT /applications/id/collaborators/username

romeovs, 2016-12-07:
I agree. I'll work on this soon.

cambierr, 2016-12-07:
Also I didn't found any doc about /applications/{id}/accesskeys/ Is this normal ?

romeovs, 2016-12-07:
So currently its:

```
GET /api/v2/applications/{id}
{ ..., "collaborators": [ { "name": "...", ...} ] }

GET /api/v2/applications/{id}/collaborators
{ "name": [ "right1", ... ], ... }

GET /api/v2/applications/{id}/collaborators/{username}
[ "right1", ... ]

PUT  /api/v2/applications/{id}/collaborators/{username}
(expects) [ "right1", ...]
```

And I'm changing it to:
```
GET /api/v2/applications/{id}
{ ..., "collaborators": [ { "name": "...", ...} ] }

GET /api/v2/applications/{id}/collaborators
[ { "name": "...", "rights": [ "right1", ... ]}, ... ]

GET /api/v2/applications/{id}/collaborators/{username}
[ "right1", ... ]

PUT  /api/v2/applications/{id}/collaborators/{username}
(expects) [ "right1", ...]
```

So basically only `GET /api/v2/applications/{id}/collaborators` changes to be more consistent.

cambierr, 2016-12-07:
Wouldn't it be possible to send the whole Collaborator object on /api/v2/applications/{id}/collaborators/{username} ? this way, we could use a single Collaborator object mapper for all calls ?

romeovs, 2016-12-07:
@cambier, yes but that's duplicating information and would result in breaking changes for our clients.
The Collaborator object contains a rights object right?

cambierr, 2016-12-07:
Right but I think that most sdk developers are using json to/from object mapping tools so it would be easier if we could use the same mapper for both /collaborators with an output like a Collaborator object array and /collaborators/id with an output being a Collaborator object alone.

romeovs, 2016-12-07:
I get what you mean, but this would mean changing all of our existing codebases, so I'm a bit wary of introducing this just to avoid another unmarshalling type.

I've no experience with how hard this is in some languages, but unmarshalling an array of strings seems easier to me than unmarshalling the Collaborator.

cambierr, 2016-12-07:
Let's keep it the way it is now then! We'll just have to add a new structure (well, not really since it's just a string array)

romeovs, 2016-12-07:
In the meantime, this stuff is implemented and tested. Just waiting for my CD script to complete so I can deploy!

johanstokking, 2016-12-08:
Agreed with making the API friendly for deserialisation. What also plays here is that when returning an object, we reserve more freedom to add other fields, and when returning the entire object (including the name), we get the original casing.

cambierr, 2016-12-09:
Also, could you add GET on /applications/{app_id}/access-keys ?

htdvisser, 2016-12-12:
@romeovs, @FokkeZB: Update on this?

romeovs, 2016-12-12:
Changes have been made and deployed

closed 2016-12-12 by romeovs