TheThingsArchive/ttn #376: KR Channel Plan

htdvisser, 2016-12-08:
**Default LoRaWAN Channels**

| Frequency | Bandwidth |
| -- | -- |
| 922.1 | 125 |
| 922.3 | 125 |
| 922.5 | 125 |

**Proposed Additional Channels (through `CFList`)**

| Frequency | Bandwidth |
| -- | -- |
| 922.7 | 125 |
| 922.9 | 125 |
| 923.1 | 125 |
| 923.3 | 125 |
| 0 | 0 |

Where the last channel could probably be `921.9` instead of _disabled_.

See also https://www.thethingsnetwork.org/forum/t/a-poly-packet-forwarder-configuration-file-for-korea/4125/4

hoonppark, 2016-12-22:
I think the above proposal is good KR920-923 channel plan.  It is better to disable the last channel instead of putting 921.9 in there.  

Maximum EIRP of all channels list above in the proposal is 14 dbm.   921.9's maximum EIRP is 10 dbm which will give us less distance coverage than 14 dbm.   It would be better to have uniform EIRP on all used channels, so users can expect a uniform distance coverage regardless which channel is used at a given moment.

What needs to be done to finalize the KR Channel Plan?  What is the procedure to finalize it?

htdvisser, 2016-12-23:
Ok. Let's try to add the proposed channel plan for Korea in our `v2.1.0` release. Please finalize the gateway configuration (https://github.com/TheThingsNetwork/gateway-conf/issues/7) for this frequency plan and submit it a Pull Request to that repository. I'll take care of the configuration on the backend side.

hoonppark, 2017-01-03:
@htdvisser,  the KR920-923 ISM band exclusively uses LBT channel access rule as stated by @tftelkamp  at https://github.com/TheThingsNetwork/gateway-conf/issues/7.

Because KR920-923 channels use LBT only, there is no need for the Fair-usage-policy or Duty-cycle limitation enforced by ETSI for KR920-923.   The ttn source code needs to be modified not to enforce the Fair-use-policy and Duty-cycle limitation for KR920-923 channel plan.

@htdvisser , Can we also include this modification in v2.1.0?

htdvisser, 2017-01-03:
The duty cycle is [only enforced for EU devices](https://github.com/TheThingsNetwork/ttn/blob/develop/core/router/downlink.go#L259-L282). The fair-usage-policy is from The Things Network and has nothing to do with local regulations. It applies to all devices on TTN. Therefore no additional changes are needed.

hoonppark, 2017-01-04:
@htdvisser, when I run my own private TTN server in Korea, I'd like to allow devices to transmit data as often as they want.  So, I'd like to remove the fair access policy (FAP) for my own private network.  In order to remove FAP on my private TTN server, which source code module should I modify?

johanstokking, 2017-01-04:
@hoonppark the FAP is not implemented yet, so nothing to delete. You'll be able to disable it for private networks without needing to change code. It will be managed by the Network Server component and enforced by the Broker.

commit 12973629 (2016-12-23): Add KR920-923 Channel Plan

Resolves #376
PR #398 (2017-01-09): Add KR920-923 Channel Plan
Resolves #376
closed 2017-01-09 by htdvisser