TheThingsArchive/ttn #378: v2 gateway name cannot have capitals in the name when creating

amkuipers, 2016-12-14:
This is a **{bug report}** for **{the Console}**.

- Explain what you want to do
Add a gateway
- Explain what steps you took
Fill in the name: Gway-Westwijk-1ch
- Explain what went wrong or what is missing
The error does not explain that capital letters are not accepted, only lowercase letters.

## Environment

- My gateway:
  - EUI: `{...}`
  - Connected to: `router.{...}.thethings.network:1700`
  - New created: https://console.thethingsnetwork.org/gateways/gateway-westwijk-1ch

johanstokking, 2016-12-15:
We don't allow capitals in the gateway IDs.

@romeovs will look into the error message if anything is not clear.

romeovs, 2016-12-15:
The error now states that the id should be lowercase.

closed 2016-12-15 by johanstokking