TheThingsArchive/ttn #384: ttnctl: applications select doesn't take ID and talks about EUI

FokkeZB, 2016-12-16:
When I run `ttnctl applications select` it doesn't allow me to pass an application ID like a previous version allowed me to pass an EUI. And when I have selected an app it talks about its EUI. Is this a legacy or as expected?

```bash
$ ttnctl applications select -h
ttnctl applications select can be used to select the application to use in next commands.

Usage:
  ttnctl applications select [flags]

Examples:
$ ttnctl applications select
  INFO Found one application "test", selecting that one.
  INFO Found one EUI "0000000000000000", selecting that one.
  INFO Updated configuration

[ edit ]

Fokkes-MacBook-Pro:docs fokkezb$ ttnctl applications select hello-worlds
  INFO Found 10 applications:

  	ID                  	Description

[ edit ]

10	hello-worlds        	Hello World App

Which one do you want to use?
Enter the number (1 - 10) > 10
  INFO Found one EUI "70B3D57EF0001B5F", selecting that one.
  INFO Updated configuration
```

## Environment

- `ttnctl version` returns: `Branch=master BuildDate=2016-12-16T12:26:13Z Commit=534943f17e00f5fde6fa56401f92b19804a9a32f Version=v2.0.1-dev`

htdvisser, 2016-12-19:
1. We should offer a non-interactive version of this command, allowing you to supply the parameters, instead of selecting them in a list.
2. The EUI is still needed, as we have to select a default EUI where devices should be added to if none is supplied by the user when creating a device.

closed 2016-12-23 by htdvisser

FokkeZB, 2016-12-23:
@htdvisser great! I'll update the quick start. Can you also update generated docs?

htdvisser, 2016-12-23:
c6747c4d2abc104d49285dbf039fac8fe0b73232

commit 958274fe (2016-12-23): Add non-interactive AppID/AppEUI selection to ttnctl

Resolves #384