TheThingsArchive/ttn #406: Include DevEUI in uplink message

johanstokking, 2017-01-11:
This is a **feature request** for **the Backend**.

I'd like to have the DevEUI as part of the uplink message. Upstream components (e.g. IoT platforms) may use the DevEUIs to look up devices and may not be aware of a custom Dev ID. Currently, the advised way is to use the DevEUI as Dev ID, but I'd like to be able to use the Dev ID for human readable reference.

htdvisser, 2017-01-12:
How would this work when we start supporting non-LoRaWAN devices? Or are we okay with being tied with LoRaWAN like this?

johanstokking, 2017-01-12:
Put it in the metadata; that's already LoRaWAN specific. Also we're not tied to the term DevEUI there; it's more like a hardware serial

PR #414 (2017-01-16): Add IDs and HardwareSerial to Uplink Messages on MQTT
Intentionally *not* using `DevEUI` so that HardwareSerial can be MAC/EUI/UUID/...

Resolves #406
commit 7f88cd3e (2017-01-16): Add HardwareSerial to MQTT Uplink

For LoRaWAN this is the DevEUI

Resolves #406
closed 2017-01-16 by johanstokking