TheThingsArchive/ttn #424: Console, migrate gateway to new packet forwarder

jpmeijers, 2017-01-20:
This is a feature request for the Console.

- Migrate a gateway from the Semtech protocol to the new packet forwarder.
- Currently you need to delete the gateway and create a new one.
- It would be nicer to just "migrate" the gateway to the new protocol/connector.
- This will allow logging the old EUI and associating it with the new ID. This is good for references purposes if old data needs to be associated with a "new" gateway.

romeovs, 2017-05-23:
Technicallyy you don't really need to d anything right? The gateway will just work when you install the new packet forwarder.

jpmeijers, 2017-05-23:
Yes, you are right. I was not aware of that when I created this issue. 

Perhaps close this issue, but put a +1 under "renaming gateway" and "change owner of gateway".

closed 2017-05-23 by romeovs