TheThingsArchive/ttn #425: Register new devices on join

johanstokking, 2017-01-20:
This is a **feature request** for **the Backend**.

Currently, only devices that are registered can be accepted when joining the network. This requires all devices to be known, including their DevEUI. This might not be practical in large scale deployments.

Therefore, it should be possible for devices to register on join. To avoid join challenges to be sent to all Handlers that registered the AppEUI with a Broker, there should be a first-come-first-serve registration of AppEUI. We might leverage the current unique registration of AppEUI/DevEUI but then with a blank DevEUI. If the Broker doesn't find a specific AppEUI/DevEUI combination for the join, it can challenge a Handler that has the AppEUI registered with a DevEUI as catchall.

## On Security

When allowing any DevEUI to join for an AppEUI and the AppKey leaks, we basically enable spoofing of existing devices and registering new devices. This means that the application needs its own mechanism to identify the device. Many IoT platforms already support this with a device registry. Also, when introducing this feature, we should inform the application owners of the potential risk.

Also, we should restrict the AppEUIs that can register devices anonymously to a range in the Foundation's EUI block. Otherwise, people can abuse The Things Network to send conflicting join accepts for any join request as AppEUIs are public. We can either configure the Brokers with an environment variable for a range.

robouden, 2017-01-20:
Happy to see the request. 
I hope it can be implemented soon.

rob

dotpy3, 2017-03-28:
This is a proposal for an implementation of this:
+ **User opt-in activation:** The user activates the option through the console, with `ttnctl`, or directly with the API.
  + The user instructs the handler to allow on-join registration for a specific `AppEUI`. The handler stores this decision in the app (through db storage, in the current implementation). An `AppKey` called `AppKey1` is either generated for those devices, or it is specified by the user.
  + The handler sends a new device registration request to its broker `B1`, for an app with a blank `DevEUI`.
  + The broker then retransmits this to the Network Server `NS1` for storage.

+ **First join request scenario:** Considering a new device, with `DevEUI` and `AppEUI`.
  + When configuring the device, the user fills in the `AppEUI` with the right `AppKey`.
  + Router transmits to brokers the activation request, with the device's correct `DevEUI` and `AppEUI`.
  + The brokers interrogate their NS - `B1` interrogates `NS1`. `NS1` looks for devices in its storage that have the same AppEUI, and that have either the same DevEUI, or an empty DevEUI. It finds a registered device with a blank `DevEUI`. It thus generates a copy of that device, but with the correct `DevEUI`. This copy, called `D1`, is processed and sent to the broker, as if it was already registered.
  + The join procedure can then continue as usual.

+ The **uplink and downlink procedure** doesn't change, given that starting from the first join, the new device has been stored as a regular device.

+ **Opt-out procedure**: When the user opt-outs of this option, the handler asks for removal of the device of the app with the empty `DevEUI`. The devices generated in the meanwhile stay in the network server's storage.

One thing to take into account, if it isn't already implemented, will be to prevent users from registering devices with empty `DevEUI`s, to prevent them from activating this option by accident.

I'm going to start on this implementation, feel free to comment if anything seems off.

rayozzie, 2017-03-28:
Hello.  I believe that what you said above does indeed address this case, but just to rephrase from the perspective of an app admin/developer.  (I get confused because "user" is something I call the person who is using our devices.):

1. The app admin/developer in this case implicitly acknowledges that they are prioritizing "ease of registration" over other factors, and implicitly acknowledges that they are going to diligently "device validity checking" at a higher level within their app.  For any device that becomes suspect at the app level, the app or its admin will use ttnctl or the api to disable that device.

2. The app admin/developer in this case explicitly indicates this by setting some checkbox in the console, ttnctl, or through the api.  I'll refer to this as the "Auto-Registration" flag.

3. Once, up-front, prior to device manufacturing, the app developer uses the ttn console to assign or fetch an AppEui+AppKey pair to be used for mass-manufactured devices.

4. Upon first and successive joins, the mass-manufactured devices present the tuple of (AppEui, AppKey, DevEui) to TTN.

5. When receiving this tuple for an unregistered device, provided that the app admin/developer had set that "Auto-Registration" flag, the device will self-register using that DevEui as the device ID.

I hope this makes sense.

dotpy3, 2017-03-29:
This makes sense, thank you for rephrasing it from this perspective. There will probably be some pedagogy to do for the app administrators in the console, so that they are aware of the good practices revolving around this functionality (an explicit guide on how to use this function to register devices, that they should disable suspect devices...).

rayozzie, 2017-03-29:
These are all very good points, Eric.

I've really just thought of the "enable/disable" feature as something that could be used by an app to put something into a questionable state while they evaluate the situation.  If a device were exhibiting DOS behavior it might be good to spare the mqqt or http integration the pain of dealing with the flood.
    
Not to get too far off-track, but the more I've thought about these issues, the more I keep coming to the conclusion that the following would be a reasonable architecture.  Forgive me if this is already what you'd been thinking about:

1.	As today, devices would be mass-produced and join OTA with a hard-wired AppEui/AppKey burned into the firmware, using HwEui or some other Eui as a unique device ID.

2.	An app written to TTN’s “suggested architectural guidelines” would dedicate an AppEui to unauthenticated devices – referred to herein as the “StagingAppEui” – and would create another AppEui for authenticated devices referred to as the “ProductionAppEui”.  Devices using the StagingAppEui (as is obvious in the UplinkMessage) might be given limited functionality as compared to those using the ProductionAppEui.

3.	When it decides to shift that device into a more “trusted” state, the app would then generate, and use a TTN API to register, an AppKey and potentially even a new DevEui, to be used with the ProductionAppEui.  The app would call ttn.RegisterDevice(ProductionAppEui, AppKey, DevEui) to prepare TTN to accept an incoming device join.

4.	Then, the app – in its own app protocol – would send the device that new tuple of (ProductionAppEui, AppKey, DevEui).  The device would store it into nvram and to be used for all future joins.

5.	At any later time, based on questionable device behavior, the app could change its mind by using ttn.EnableDevice(AppEui, AppKey, DevEui, fEnabled) or ttn.DeregisterDevice(AppEui, AppKey, DevEui).  If behavior ever warranted, it could even send it a new re-registration tuple to move it to a “QuarrantineAppEui”.

Yes, if someone steals a device they can spoof themselves by using that device's identity tuple.  But this is an unavoidable attack vector: even with an integrated secure element if someone steals a device they can desolder and repurpose its microcontroller while maintaining authentication.
 
Just a thought.

dotpy3, 2017-03-30:
I understand better what you meant regarding the "enable/disable" feature, and it makes sense as a first reaction feature before evaluating the situation for the app as a whole. @htdvisser @johanstokking, do you have any thoughts on this feature?

After reading your architecture proposition carefully, it seems to me like an interesting implementation architecture of a trust system in case of on-join registration. A such registration process would allow the app administrator to keep track of trusted devices based on their `AppEUI` - making a clear distinction between devices that have established a certain level of trust, and others. Elements that would require specific attention during this process, in my opinion, would be:

+ Using efficient encryption during the 3rd step, to avoid interception of the new AppKey.
+ An app with questionable behaviour might not execute the re-registration step evoked in the 5th step toward the `QuarantineAppEUI`, or even back to the `StagingAppEUI`. This should be assumed by the app administrator and developers, and kept track of - with a device registry, for example.

rayozzie, 2017-03-30:
Agreed. 

I'm thinking that this scheme would be even more useful if it went something like this:

A. Whenever creating a new app on the service, TTN would allow me to identify the app as an "Automatic Device Registration" app.  As part of this, TTN would secretly create, for its own use, a special "AutoAppKey" public/private key pair.

B. When first joining an app with that app's AppKey, the device would specify an AppKey of 0. This would tell TTN that the app is in an automatic registration mode.  TTN would auto-register the device, but would add a new flag to The UplinkMessage data structure called "Authenticated Device", set to false. All inbound traffic from any device presenting an AppKey of 0 on join would be perfectly valid but would have this flag set to false in uplink messages.

B. At its option, the app could then ask TTN, at its API, to generate: DevAppKey := TTN.GenerateDevAppKey(myApp, DevEui)
This app key would be generated by TTN by internally doing Encrypt(myApp.AutoAppKey.PrivateKey, Marshal(myApp.AppKey, DevEui, RandomSalt))

C. The app would then return this DevAppKey to the device on the downlink channel.  The device would save this in NVRAM.

D. The device would then rejoin TTN, using the same AppKey and DevEui that it had used previously, but using this DevAppKey instead of 0.

E. Any server throughout or even outside TTN, of course, can then validate the DevAppKey on the rejoin by using its own Decrypt(myApp.AutoAppKey.PublicKey, DevEui).  After the rejoin with this AppKey, the "AuthenticatedDevice" flag would be set to true.

F. If you were to support an Enable and Disable feature to an app, again it would be done as TTN.EnableDevice(myApp.AppKey, DevEui, fEnable).  Deregistration need not be implemented in TTN as this is better handled at the app level, given that it could always just auto-reregister.

The advantage to this scheme, as opposed to the one that I suggested above, is that it only requires a single AppKey to be managed by the app, has less network traffic, and is secure.

johanstokking, 2017-04-03:
Thanks for the input gentlemen! If I understand correctly, there are two separate issues here; automatically registering new devices on join, and sending a new key to a device. I think we can implement the former simply, and see what the LoRa Alliance TC thinks about the latter.

For the simple implementation of this issue, I agree with the user flow as described by @rayozzie https://github.com/TheThingsNetwork/ttn/issues/425#issuecomment-289842102

In TTN, this would be implemented as follows:

- The Handler registers the combination AppEUI + empty DevEUI on the Broker/NS. The combination of AppEUI and DevEUI is unique, so registering with an empty DevEUI is basically claiming the AppEUI. Remember you can add any AppEUI to your application on a first-come-first-serve basis

- The Handler registers the combination of AppEUI + empty DevEUI in its own registry, with the common AppKey

- Whenever a Broker receives a join, it first queries the NS whether there's a device with the AppEUI and DevEUI present. If so, follow current flow. If not, see if there's a registration with AppEUI and blank DevEUI present. If so, forward to the Handler as normal (including key check)

- When the Handler receives a join with an empty DevEUI and it has the right key, it registers a new device on the Broker with the DevEUI, and registers the new device in its own registry with the DevEUI and AppKey, and completes the join as normal

closed 2017-05-03 by johanstokking

vanthome, 2018-03-01:
Hi, I just tried to OTA my device with an DEV EUI of all zeros. Unfortunately it does not seem to work. So should this actually already work? If, so, it would be nice to mention it also here: https://www.thethingsnetwork.org/docs/devices/registration.html

htdvisser, 2018-03-02:
An all-zero DevEUI is not a valid EUI (by definition), so you can not use that

vanthome, 2018-03-02:
ok, what would I need to use then to achieve the behaviour discussed here?

johanstokking, 2018-03-04:
@vanthome please use `ttnctl devices register on-join ...`:

```
$ ttnctl devices register on-join --help
ttnctl devices register on-join can be used to register a device template for on-join registrations.

Usage:
  ttnctl devices register on-join [Device ID Prefix] [AppKey] [flags]
```

neolinux, 2018-07-24:
Hello @johanstokking ,

this doesen't seems to work:

PS C:\Users\****\Downloads> .\ttnctl-windows-amd64.exe devices register on-join demo B78906*********
CA1FDD9**********
  INFO Using Application                        AppEUI=70B3D57****** AppID=*****
  WARN On-join registration does not follow LoRaWAN best practices. Unless you implement additional security
  WARN measures, it may corrupt your entire application if even one of your devices gets compromised.
  INFO Discovering Handler...                   Handler=ttn-handler-eu
  INFO Connecting with Handler...               Handler=eu.thethings.network:1904
 FATAL Could not register Device                error=Could not set device on Handler: Invalid Device: Invalid Device: Invalid LoRaWANDevice: DevEUI not valid: can not be empty source=Wrap: /go/src/github.com/TheThingsNetwork/ttn/utils/errors/errors.go:222
PS C:\Users\****\Downloads>

Why?

Thank You.

htdvisser, 2018-07-24:
Hi @neolinux, you already reported this in #702.

PR #585 (2017-05-03): Register devices on join
This PR adds functionality for registering devices when they join the network.

The steps to do this are as follows:

1. `ttnctl applications select [AppID] [AppEUI]`
2. `ttnctl devices register on-join [Device ID Prefix] [AppKey]`
    - This will add an access key to your application if needed
3. Join a device with AppEUI and AppKey - Any DevEUI is accepted
4. The device is registered, its DevID will be `Device ID Prefix`-`DevEUI`

Closes #425