TheThingsArchive/ttn #427: Gateway bridge support lorank format

jpmeijers, 2017-01-23:
This is a feature request for the Backend/the NOC.

The gateway bridge currently does not process status packets from Lorank8 gateways when they are using their default configuration. 

Example status packet from Lorank8:
`{"time":"2017-01-19T18:02:38Z","dev":{"id":"1dee039aac75c307","lat":52.2388,"lon":6.8551,"alt":6,"up":93,"gps":"fake","pfrm":"Lorank","email":"j.vaningenschenau@utwente.nl","desc":"TTN Enschede macrocell gateway"},"prf":{"up_rf":0,"up_srv_dg":[0,1],"dw_srv_hb":[0.8333333333,1],"dw_srv_dg":[0,0],"dw_rf":[0.8333333333,1],"dw_bcn":0}}`

This format is slightly different than the Semtech protocol.

In the configuration of the Lorank8 one can change `"stat_format": "idee_concise",` to `"stat_format": "semtech",` to change to the Semtech protocol. `idee_concise`, which is the format for the example above, is the default.

A little background info:

> vlaming
> It the subnet your lorank is located use a browser to go to http://lorank8/ or http://[IP number]/ and you should see the control page of the device. Go to the Admin tab, Stop the concentrator (give it 10 sec). Choose the  Status format Semtech below that and Start the concentrator again. Thats all.

Please support the `idee_concise` format as well so that we can see the Locations of these Lorank8 gateways on the NOC.

johanstokking, 2017-01-23:
Great community PR candidate! cc @jpmeijers @devlaam

htdvisser, 2017-01-23:
Probably a good idea to do that upstream: https://github.com/brocaar/lora-gateway-bridge

jpmeijers, 2017-01-27:
From a reply on the upstream repo, and after a discussion on the TTN Slack, I came to the conclusion that this is a WONT_FIX. Creating a new issue for a workaround.

closed 2017-01-27 by jpmeijers