TheThingsArchive/ttn #437: Make use of the full width of the screen

jwillemsen, 2017-01-30:
This is a feature request for website.

I noticed when looking at the application data on the website there is just a lot of gray space on the left and right which is not used by the website. Especially when also having a payload function not much data can be shown and there is no feature to scroll left/right.

Also look at the buttons for uplink/downlink/activation, not always clear for someone who is color blind like me when they are enabled/disabled.

jwillemsen, 2017-01-30:
Attached screenshot, I can scroll

jwillemsen, 2017-01-30:
[image]

htdvisser, 2017-06-28:
- The width of the console was changed some time ago and should be a lot better now.
- The contrast of the filter states was also improved, please reopen if not good enough.

htdvisser, 2017-06-28:
Executing the following javascript in the browser console will make the UI fill the entire window:
```js
var s = document.createElement("style");
s.appendChild(document.createTextNode("main,main>div,nav>div{max-width: 100% !important}"));
document.head.appendChild(s);
```

closed 2017-06-28 by htdvisser