TheThingsArchive/ttn #446: Make De-Duplication "horizontally scalable"

htdvisser, 2017-02-01:


cambierr, 2017-07-19:
I'm interested by this one.

The way I see it is by using redis and something like `append` to store all messages having the same key (keeping the way it's currently computed in the `broker`) in the same key-value pair.

Once the timeout is reached, the complete value (containing one to n messages) is being retrieved and computation is made to determine which broker instance will process the message (using the first that received it for instance) and finally forward it to the good handler.

What's your opinion on this ?

htdvisser, 2017-07-20:
I was thinking about something that even is a bit more scalable than using one, central, big, shared memory system (Redis).
[image]

cambierr, 2017-07-20:
I see (and like) the idea but what if the deduplicator for hash H1 goes down ?

I agree on the SPOF/SPOP that is redis, but we could imagine another high-performance and scalable database/cache like CouchBase which offers the same kind of features plus persistence, native clustering/HA/LB & super-easy administration ?

Plus, CouchBase provides atomic json edits which we could use to create a document being an array of uplink messages before deduplication occurs ?

htdvisser, 2017-07-21:
My reservations are more about the fact that we're going to use an external **store** (meant for storing stuff) for de-duplication. 

Maybe we're also overthinking this a bit. We already shard messages over different Brokers based on the DevAddr of the message, so maybe we should just keep it simple and use the existing sharding mechanism instead of coming up with yet another microservice.

Let's also keep processing time in mind. Any additional component (datastore or dedicated deduplicator) will add several milliseconds to the processing time, which shouldn't be necessary.

cambierr, 2017-07-21:
I agree on the overthinking part.

My only concern about the sharding is what if the shard goes down ?

closed 2018-10-05 by htdvisser