TheThingsArchive/ttn #450: Fix the typo on ttnctl applications command; missing "applications" word

edwindvinas, 2017-02-02:
This is a typo correction for ttnctl applications command.

Impacted entities:
ttn\ttnctl\cmd\applications_register.go
ttnctl\cmd\docs\README.md 

Please correct typo as shown below for the "Long: " description. "ttnctl applications register" should be the correct instead of "ttnctl register".

* FILE: ttn\ttnctl\cmd\applications_register.go
var applicationsRegisterCmd = &cobra.Command{                                        
 Use:   "register",                                                                  
 Short: "Register this application with the handler",                                
 Long:  `ttnctl register can be used to register this application with the handler.`,
>>>> Long:  `ttnctl applications register can be used to register this application with the handler.`,
 Example: `$ ttnctl applications register                                            
  INFO Discovering Handler...                                                        
  INFO Connecting with Handler...                                                    
  INFO Registered application                   AppID=test                           
`,                                                                                   

* FILE: ttnctl\cmd\docs\README.md 
### ttnctl applications register                                          
                                                                          
ttnctl register can be used to register this application with the handler.
>>>> ttnctl applications register can be used to register this application with the handler.                                                                          
**Usage:** `ttnctl applications register`

htdvisser, 2017-02-03:
Thanks for your report. Feel free to fix this and submit a Pull Request.

closed 2017-02-03 by htdvisser

johanstokking, 2017-02-03:
@edwindvinas let me fix that for you

[image]

PR #456 (2017-02-03): Fixed ttnctl cmd typo
Resolves #450