TheThingsArchive/ttn #460: Registering Application - handler dropdown selection ignores mouse

dlarue, 2017-02-05:
This is a **{bug report}** for **{the Console}**.

- Explain what you want to do
Dropdown menu to select the US-West handler ignores the mouse, must use keyboard arrow keys.

- Explain what steps you took
Select Register Application, go to the "Handler registration" section and try to use your mouse to select the us-west handler. Clicks do not result in the us-west handler selection.  But use the keyboard arrow key to scroll down to select the us-west handler and hit enter key and it registers.

- Explain what went wrong or what is missing
Appears to a missing mouse input handler.

## Environment
TTN V2 backend in CONSOLE

dlarue, 2017-02-05:
I just noticed that if I use the keyboard the first time to select the other handler, subsequent uses of the mouse will work.  I still get an error after trying to register the application stating that the handler was not created.  This seems to be false since attendees of my workshop had been able to create devices and get data from physical devices with the us-west handler selected.

avbentem, 2017-02-06:
I recall seeing kind of the same behavior when editing gateways: I could not select most fields such as the gateway type (I'm not sure I tried going down with the keyboard in the dropdown) but I could start typing it myself and then select it with the keyboard indeed. 

I think this is only true for fields that also allow for typing one's own value.

johanstokking, 2017-02-06:
Confirmed, will be fixed

romeovs, 2017-02-06:
Updated the console to fix this issue, please confirm.

htdvisser, 2017-02-06:
Looks fixed to me :)

dlarue, 2017-02-06:
Not fixed in the live environment.  The US and EU handlers are below view and require scrolling the drop down and those two items do not take the mouse clicks.

dlarue, 2017-02-06:
It turns out that even the handler selection in the CONSOLE global Settings page operates the same way. No mouse clicks for handlers below view in the dropdown list.

romeovs, 2017-02-06:
I can reproduce even after the fix. It seems to be a timing issue, I'll investigate.

@diarue All of these dropdowns share the same component so it makes sense that they suffer from the same issue.

dlarue, 2017-02-06:
and nicely be fixed by a single repair. :)

closed 2017-03-02 by johanstokking