TheThingsArchive/ttn #461: unregister then re-register of application handler only allows selection of eu handler

dlarue, 2017-02-05:
This is a **{bug report}** for **{the Console}**.

- Explain what you want to do
After registering the us-west handler I was presented with a notification that the handler was NOT registered.  So I went back into the Application settings and clicked unregister and was only presented with the option of registering with the eu handler.

- Explain what steps you took
- Explain what went wrong or what is missing
None of the other handlers options are available and deleting of the application is the only option.

## Environment
TTN V2 backend CONSOLE

dlarue, 2017-02-05:
After deleting the application to recreate it with the US-West handler, I was unable to use the same name and the system said there was already an application with that name.

romeovs, 2017-02-06:
This is a UX issue. The handler that you could register to was a console global setting (Settings > Default Handler).

I've updated the console make this more straightforward.

Re-creating applications with a same name as before is not yet possible in the console, but it's a planned feature.

dlarue, 2017-02-06:
Ah, I see now that there's a settings page for CONSOLE. I've updated that. Thanks.  FYI, the bug I reported about not being able to mouse click the handlers I have to scroll to(ex: us-west). I have to use keyboard arrow keys and the enter key to select those items in the dropdown list.

romeovs, 2017-02-06:
@diarue Okay, can I consider this closed? I'm working on the other issue.

dlarue, 2017-02-06:
sure, it's a UX quirk but workable.

romeovs, 2017-02-06:
@dlarue. But do you agree that the quirk has been resolved by adding the dropdown to application settings (regardless of whether or not #460 is fixed).

dlarue, 2017-02-06:
yes

closed 2017-02-06 by dlarue