TheThingsArchive/ttn #477: Send device straight to a higher SF on join if possible

johanstokking, 2017-02-16:
This is a **feature request** for **the Backend**.

I want to send devices that are joining straight away to a higher SF if possible. This is to avoid that SF12 devices stay too long there.

htdvisser, 2017-02-20:
Unfortunately there is no "clean" way to do this under the current LoRaWAN specification. For devices that use ADR, it might be possible to make them increase their data rate faster. For non-ADR devices, the only option I see is to send a MAC command to completely disable SF12 (possibly re-enabling it later), but I don't think this is the most elegant method. Maybe @tftelkamp can think of something?

johanstokking, 2017-02-20:
Hardcoded SF12 devices that do not have ADR enabled should not be permitted anyway. What is allowed though, are devices that send data on different SF's without having ADR enabled.

This issue affects ADR enabled devices mostly. The signal of the activation request may indicate that we can move to a faster data rate, straight to SF10 or SF9, possibly even higher.

htdvisser, 2018-05-02:
- Some device implementations are indeed a bit disappointing. The recommended flow has always been to start attempting to join at the best data rate (SF7) and fall back to less efficient data rates if that doesn't work.
- With US915 devices we **have to** send an ADR command in order to configure the channels it can use. With LoRaWAN 1.1 this becomes part of the JoinAccept message, and we'll be able to separate channel setup from ADR.
- We really need to collect some data before we can send ADR commands to move the device to a different data rate. If we send such ADR commands with insufficient data, we'll have a pretty big risk of "disabling" a device with inappropriate ADR settings. We already saw this happening to some US915 devices, so we added an extra margin to the initial ADR command for those.

We could consider sending an "initial" ADR command to EU868 devices, similar to what we do with the US915 devices. Not sure if this should still go to our v2 backend or that we'll push it to v3.

htdvisser, 2021-06-03:
This won't be implemented in V2 anymore.

V3 is more aggressive with ADR, and uses an extra "safety margin" when it doesn't have enough measurements yet.

closed 2021-06-03 by htdvisser