TheThingsArchive/ttn #478: Set gateways "Antenna Placement" does not work

h0l0gram, 2017-02-18:
This is a **{bug report}** for **{the TTN Console}**.

- I configure a gateway's "location" and set its "Antenna Placement" to outdoor. Then save the settings.
[image]

- Problem:
  - When I then see the gateways "Antenna Placement" it still shows "(?)"
[image]

- My gateway:
  - EUI: `{eui-b827ebfffe0ca4a4}`

htdvisser, 2017-02-20:
I can confirm this issue. Regardless the "antenna placement" setting, the overview does not show the placement.

romeovs, 2017-02-20:
Fixed and deployed!

closed 2017-02-20 by romeovs

h0l0gram, 2017-02-20:
Confirmed. Its working now! Thanks guys!