TheThingsArchive/ttn #479: Increase time that router schedules downlink to gateway

johanstokking, 2017-02-20:
This is a **feature request** for **the Backend**.

Currently, the router schedules downlink messages 100ms before they should be transmitted to the gateway. This is too short for many types of gateway connections (e.g. DSL and cellular) and/or remote locations.

If there's RTT available, we should account for that, but if there isn't, the default should be 1 second. This means that for gateways running older packet forwarders that they will more often overwrite downlink messages, but that is always better than consistently not transmitting downlink at all.

htdvisser, 2017-02-20:
To avoid double work, this should be implemented together with #420.

htdvisser, 2017-03-20:
Due to delays in #420, the schedule time was doubled 10 days ago in 1bb19bf818fb4d542b31290076cb888e1b899b98

closed 2017-03-20 by htdvisser