TheThingsArchive/ttn #482: Text-Widget in Windows Firefox is vertical mis-aligned

h0l0gram, 2017-02-20:
This is a **bug report** for **the Console**.

Using firefox **51.0.1 (32-bit)** on Windows 7. The "text widget" does not show the text in a readable manner. The widget shows a scroll bar and the text is vertical not well aligned.

**Description** is broken (most other text widgets on other settings are broken as well)
**Frequency** Plan is ok

[image]

avbentem, 2017-02-20:
Same on latest Firefox on a Mac, while all is fine in latest Chrome on the same Mac.

romeovs, 2017-02-21:
@avbentem @h0l0gram I deployed a fix, can you confirm it works and close the issue accordingly?

avbentem, 2017-02-21:
Nice, @romeovs, the alignment looks okay now in Firefox 51.0.1 64 bit on a Mac. (And also in my Chrome and Safari.)

However, **I do not see a cursor when the form field is focused in Firefox**. (I guess I didn't see it yesterday either.) This makes it hard to guess that one can type whatever one wants. I can move around in the field, like I can go back a few characters and then type something in the middle of any existing value. So, the text I type appears just fine, but I cannot see where I'm about to type.

Also, any text selection is not visible. I can select a few characters while holding Shift and use the arrow keys, and then hitting Delete will delete only the invisible selection. But I cannot see what I selected.

All is fine in true text fields, such as the gateway's description.

(In Chrome and Safari, I do see the cursor when the field has focus, and I also see selected text.)

h0l0gram, 2017-02-21:
Thanks @romeovs looks better now on Firefox on Linux (51.0.1 (64-bit)):

- [x] While the vertical alignment is fixed,  
- [ ] there are vertical and horizontal **scroll bars** visible, despite the list empty
- [ ] As stated above **no cursor** is visible in "text-widget".

[image]

I'll report Firefox on Windows tonight.

romeovs, 2017-02-23:
Deployed some further fixes, can you confirm these work on Linux?

h0l0gram, 2017-02-23:
Great work @romeovs! Looks very nice on Firefox on Linux 51.0.1 (64-bit).

- [x] While the vertical alignment is fixed,
- [x] there are vertical and horizontal scroll bars visible, despite the list empty
- [x] As stated above no cursor is visible in "text-widget"

avbentem, 2017-02-23:
Same good results in Firefox on a Mac.

h0l0gram, 2017-02-23:
And also looks good on Firefox on Windows 51.0.1 (32-bit).

closed 2017-02-23 by h0l0gram