TheThingsArchive/ttn #485: Publish MAC commands

johanstokking, 2017-02-22:
This is a *feature request** for **the Backend**.

Publish the MAC commands, not necessarily all the parameters but most importantly the commands sent, as part of metadata of uplink and downlink messages. This enables insight in, for example, why downlink has been scheduled and sent.

dotpy3, 2017-03-10:
Regarding this issue, can we specify the scope of "publish"? Is it about publishing the MAC commands to the NOC, or is it about publishing the MAC commands to the MQTT broker (which seems unlikely)?

htdvisser, 2017-03-10:
The payloads of messages are already sent to the NOC, this includes MAC commands, so I guess @johanstokking wants something on the Handler MQTT. This might not be as simple as it sounds, because (for example) for the LinkAdrReq, the NetworkServer only adds it on the Downlink, so this won't even be seen by the Handler.

johanstokking, 2017-03-10:
OK, for this we need the NOC, just as for gateway traffic

dotpy3, 2017-04-13:
These are the solutions I can imagine to this problem:

The first one is to ignore flat out the MAC commands that the handler can't see, but it sort of kills the purpose of the feature.
The other solutions involve other components, that will have to store and/or to retransmit the MAC commands that the handler can't see:
+ One would be storing these MAC commands in a struct, called e.g. `NetworkMetadata`. This struct could contain information that needs to be transmitted across the network (in this case, to a handler). This object could be either transmitted by the broker to the handler synchronously, as soon as computation of the downlink is done, or on an event related to the device/the app (e.g. as soon as the same device sends another uplink).
+ Another, as evoked, would be making the transmission through the NOC. I don't know if the handler already subscribes to an exchange with the NOC, and if not, do we want it? @rvolosatovs

johanstokking, 2017-04-14:
I would suggest it comes from the NOC and don't bother too much getting it in the Handler. It's not supposed to be on the MQTT/AMQP, but rather present in a trace log, which is what the NOC is for to provide.

htdvisser, 2017-04-21:
Ok, in that case, this is already implemented.

closed 2017-04-21 by htdvisser