TheThingsArchive/ttn #489: Frequency Plan for Japan

htdvisser, 2017-02-27:
**⚠️ this issue is incomplete**

This is a proposal for the AS923 channel plan to be used in Japan. I hereby invite all our Japanese communities to participate in the specification.

The allocated range for unlicensed use is 916.7 - 923.5 MHz (see [this doc](http://www.gs1.org/docs/epc/uhf_regulations.pdf))

**Default LoRaWAN Channels 1-2**

| Frequency | Bandwidth |
| -- | -- |
| 923.2 | 125 |
| 923.4 | 125 |

**Proposed Channels 3-7 (with `CFList`)**

| Frequency | Bandwidth |
| -- | -- |
| - | 125 |
| - | 125 |
| - | 125 |
| - | 125 |
| - | 125 |

**Proposed Channel 8-10 (with MAC command)**

| Frequency | Bandwidth |
| -- | -- |
| - | 125 |
| - | 250 |
| - | FSK |

htdvisser, 2017-02-27:
There are a couple of countries (Hong Kong, Singapore, Thailand, Vietnam) that can "choose" between the upper (#491) and lower (#489) part of the AS923 band. The join frequencies overlap, so OTAA devices will just receive the correct channel plan based on gateways in the area, but for good coverage, it's best to all use the same frequency plan in a certain region.

Based on geography, I think it would be best to have Singapore together with Malaysia, to have Vietnam and the Northern part of Thailand together with Cambodia and Laos, and the Southern part of Thailand with Malaysia.

kajilemon, 2017-03-01:
Hello @htdvisser,
Appreciate for your proposal for Asian-country, I comment for Japan's regulation.
Your proposal for <=923 is feasible for Japan's radio law, but require below conditions.

+ The center frequecncy of 250kHz bandwidth channel should be 923.3Mhz

And also advise you some additional condtion in Japan under below situation.

In case of antenna power is between >1mW and <=20mW and center frequency is between >=920.6MHz and <=923.4MHz with 200kHz bandwidth.
 
+ Carrier Sense (LBT) time is 5msec or more
+ Sending duration (DwellTime?) is 4sec
+ Pause duration is 50msec
+ The sum of emission time per arbitrary one hour (DutyCycle?) is no limit

htdvisser, 2017-03-01:
Thanks for participating, @kajilemon!

If we move the BW250 channel to 923.3, we should also do this for the >=923 to make the configuration of ABP devices less complicated. @tftelkamp: Would moving the BW250 channel to 923.3 cause any problems?

- LBT is something that is not part of the channel plan, but should be implemented in the gateways and nodes. I'll do my best to make sure to copy your point to the appropriate issue when we create it.
- I was under the impression the DwellTime was 400ms instead of 4s. Could you check this?

kajilemon, 2017-03-01:
Thanks, @htdvisser.
I understand in the same way for LBT, LBT is not something that NS can care about.
As for DwellTime, default in AS923 is 400msec , but maximum 4sec is allowd under these typical situation in Japan.

1mW < antenna power <= 20mW
&&
920.6MHz <= center frequency <= 923.4MHz

tftelkamp, 2017-03-02:
@kajilemon, thanks for your feedback on this issue!

In the 1.0.2 LoRaWAN Regional Parameters from the LoRa Alliance Japan is listed as '920-928 MHz' ISM band. Is the part above 923.4 Mhz not available, or do other regulations apply?

Would you be able to provide us with a complete overview (or reference) of the 923Mhz ISM band regulations in Japan?

kajilemon, 2017-03-03:
Hi @tftelkamp, Above 923.4 Mhz is also available in Japan.
The chart below is quoted from full regulation document which is downloadable from http://www.arib.or.jp/english/html/overview/doc/5-STD-T108v1_0-E1.pdf

| CH | Freq |
|:---|:---|
| CH24 | 920.6 |
| CH25 | 920.8 |
| CH32 | 922.2 |
| CH33 | 922.4 |
| CH37 | 923.2 |
| CH38 | 923.4 |
| CH61 | 928.0 |

(1) Antenna Power <= 20mW
[image]

(2) Antenna Power <=250mW
[image]

tftelkamp, 2017-03-03:
Thanks for the valuable information, @kajilemon.

923.3 for the 250kHz channel is fine, although we could consider to move it to 922.1 to make it not overlap with the join channels (and RX2 downlink).

FSK at 921.8 Mhz would still fit all channels in the 2x 1Mhz bandwidth of the SX1257 front-end (center frequencies 923.0 and 922.0).

htdvisser, 2017-03-03:
@tftelkamp: so you propose to put the BW250 at 922.1 and FSK at 921.8?

tftelkamp, 2017-03-07:
@htdvisser yes, but it would be great if @kajilemon could verify if this does not create any problem.

kajilemon, 2017-03-07:
@tftelkamp , BW250 at 922.1 will use 921.9MHz<->922.3MHz of 400kHz width (200kHz x2ch) in Japan's regulation. This will overlap to proposed BW125 at 922.2 channel.

tftelkamp, 2017-03-07:
@kajilemon that's fine, we have this overlap in the EU868 plan as well.

kajilemon, 2017-03-07:
@tftelkamp , thanks.

closed 2017-03-13 by htdvisser

shmrymbd, 2017-03-31:
Malaysia -  919-923MHz for SRD specifically RFID.

PR #511 (2017-03-13): Add AS923 Channel plans
This PR adds the AS920_923 and AS923_925 channel plans to the `band` package.

Additionally, this PR adds a "cache" of Frequency plans, so that we can just copy it instead of regenerating on every uplink.

This PR also prevents downlink responses and join accepts to messages on illegal uplink frequencies.

Resolves #489
Resolves #491
commit e1e49426 (2017-03-13): Add AS923 Channel plans

Resolves #489
Resolves #491

Additionally, this commit adds a "cache" of Frequency plans, so that we can just copy it instead of regenerating on every uplink.

This commit also prevents downlink and activations on illegal uplink frequencies.