TheThingsArchive/ttn #491: Frequency plan for AS923 (>=923)

htdvisser, 2017-02-27:
**⚠️ this issue is incomplete**

This is a proposal for the AS923 channel plan to be used in countries that allow the range 923-925 MHz: Brunei, Cambodia, Hong Kong, Indonesia, Laos, New Zealand, Singapore, Taiwan, Thailand, Vietnam

**Default LoRaWAN Channels 1-2**

| Frequency | Bandwidth |
| -- | -- |
| 923.2 | 125 |
| 923.4 | 125 |

**Proposed Channels 3-7 (with `CFList`)**

| Frequency | Bandwidth |
| -- | -- |
| _todo_ | 125 |
| _todo_ | 125 |
| _todo_ | 125 |
| _todo_ | 125 |
| _todo_ | 125 |

**Proposed Channel 8-10 (with MAC command)**

| Frequency | Bandwidth |
| -- | -- |
| _todo_ | 125 |
| _todo_ | 250 |
| _todo_ | FSK |

htdvisser, 2017-02-27:
There are a couple of countries (Hong Kong, Singapore, Thailand, Vietnam) that can "choose" between the upper (#491) and lower (#489) part of the AS923 band. The join frequencies overlap, so OTAA devices will just receive the correct channel plan based on gateways in the area, but for good coverage, it's best to all use the same frequency plan in a certain region.

Based on geography, I think it would be best to have Singapore together with Malaysia, to have Vietnam and the Northern part of Thailand together with Cambodia and Laos, and the Southern part of Thailand with Malaysia.

closed 2017-03-13 by htdvisser

terrymulijana, 2017-03-15:
hi, 
923-926 ( 500mW - 2 Watt ) used for RFID in Indonesia : 
http://www.rfid-indonesia.com/images/Alokasi.Frekwensi.RFID-UHF.EPC.Gen2.Indonesia.jpg

as they've stated beyond those power output will be treated like radio transmitter, can we go with that frequency still ?

thanks

htdvisser, 2017-03-15:
As far as I know, we're well under 500mW

terrymulijana, 2017-03-15:
so you're saying I can use RFID frequency license to actually use LoraWAN ?

andriyadi, 2017-07-16:
Hi @terrymulijana nice to find Indonesian here.
The draft for LoRa band in Indonesia is 919-923MHz, similar to Malaysia. But it's not final.
Btw, I'm not representing the regulator, only happen to have friends in there.

daffodilistic, 2017-07-23:
Hi,

For Singapore, I'm not sure if this document from the regulator would help, but here it is:
https://www.imda.gov.sg/~/media/imda/files/regulation%20licensing%20and%20consultations/frameworks%20and%20policies/spectrum%20management%20and%20coordination/spectrummgmthb.pdf?la=en

It seems like 923-925MHz should be clear for use?

rizkyarlin, 2018-07-09:
@terrymulijana 

sir, can you send me the url to that regulation?

PR #511 (2017-03-13): Add AS923 Channel plans
This PR adds the AS920_923 and AS923_925 channel plans to the `band` package.

Additionally, this PR adds a "cache" of Frequency plans, so that we can just copy it instead of regenerating on every uplink.

This PR also prevents downlink responses and join accepts to messages on illegal uplink frequencies.

Resolves #489
Resolves #491
commit e1e49426 (2017-03-13): Add AS923 Channel plans

Resolves #489
Resolves #491

Additionally, this commit adds a "cache" of Frequency plans, so that we can just copy it instead of regenerating on every uplink.

This commit also prevents downlink and activations on illegal uplink frequencies.