TheThingsArchive/ttn #496: Expose context in handler client

romeovs, 2017-02-28:
Currently there's no way to cancel or set a deadline on a request to the handler, using the handler manager client.

htdvisser, 2017-03-01:
I think changing the api functions wouldn't be so bad. That way we also follow the Go guidelines: _"Do not store Contexts inside a struct type; instead, pass a Context explicitly to each function that needs it. The Context should be the first parameter, typically named ctx"_ (see https://golang.org/pkg/context/)

htdvisser, 2017-03-01:
Additionally, if you pass a `context.WithDeadline(time.Time)` to `handler.NewManagerClient`, all derived contexts will have the exact same deadline, and all requests will timeout at exactly the same time.

romeovs, 2017-03-01:
Yes, let's follow the go guidelines.
I'm reopening because this hasn't actually been resolved.

htdvisser, 2018-10-05:
Won't fix this for v2

closed 2018-10-05 by htdvisser