TheThingsArchive/ttn #507: Allow key/value pairs as sensor/node metadata

dakoller, 2017-03-06:
This is a feature request for the Backend.

- Explain what you want to do

When setting up larger usecase, you might end up with different sensors used (which send into the same application), where you want to store e.g. some information about how they are deployed. (e.g. in which height, indoor/outdoor, sensor model, software version of the node)

This information should available to everybody, who listens to the event stream (console, mqtt...), and different people might - based on these metadata - do different calculations with the data.

This key/value dictionary should be read & writeable via API. Additionally each data record, which is outputted via console, mqtt etc. should have the key/value dictionary as additonal information.
 

- Explain what steps you took
- Explain what went wrong or what is missing

## Environment

- `ttn version` returns: `Commit={...}`
- `ttnctl version` returns: `Commit={...}`
- My application:
  - App ID: `{...}`
- My gateway:
  - EUI: `{...}`
  - Connected to: `router.{...}.thethings.network:1700`
- My node:
  - Device ID: `{...}`
  - AppEUI: `{...}`
  - DevEUI: `{...}`
- MQTT is connected to: `{...}.thethings.network:{1883/8883}`
- I am using the `{...}` library for the `{...}` programming language

johanstokking, 2017-04-10:
Device attributes as a key/value dictionary can indeed be useful.

A challenge here is to distinguish between generic attributes and custom attributes. The generic attributes are in fact the ones you mention; location, placement, manufacturer, model, version.

There are three possible implementations; 1) adding these as builtin device attributes, 2) introduce key/value attributes with reserved keys and custom keys, or 3) introduce key/value attributes with only custom keys (e.g. don't add UI/UX support for the fields you mentioned in TTN).

I would suggest option 2.

Sypheos, 2017-04-19:
Option 1 seem more fitting to me.

Though it will be useful, this data should be kept to the minimum.
Adding information such as location, software version, sensor model fit TTN purpose and are indeed useful to monitor the health of a device, but data such as indoor/outdoor seems to be irrelevant here. They are applications oriented data and as such should be stored somewhere else or sent by the device himself.

When setting up large use case TTN allow simple communication with devices and monitoring. Data processing, storage, business intelligence will be handled by an integration or by your own solution. There the rest of metadata will be readily available or pulled from a service or database.

dakoller, 2017-04-23:
I'd vote for option 2) , because option 1) requires you to know all potential keys/values, which later can be applied.
This limits the usecases, which other people might want to do with TTN.
In my understanding the solution should allow to stick all sensor-specific metadata to the node object in TTN.

Sypheos, 2017-04-25:
We will move on with option 2.

Here is the plan:
1. Add only reserved field, they will be generic data that can be filled by TTN or the user. Most likely to be under the form 'ttn_..."
2. Customs keys will be added later but limited. Having a few keys can indeed prove very useful but providing more is not TTN job. That's why we have integration with more complex platform such as Azure, AWS and other IoT broker.

I will open issues for each development phase and reference them here.

Sypheos, 2017-04-28:
@dakoller What type of data you want (location, sensor model, firmware version...) ? 
There will be very few of keys available (white listed) in the first version. So if you can tell us the one you want, we can add them now.

johanstokking, 2017-04-29:
I would suggest:

- Brand
- Model
- Version
- Antenna
- LoRaWAN module

We already have the location

htdvisser, 2017-06-22:
Closed by #605. Will be included in next release

closed 2017-06-22 by htdvisser