TheThingsArchive/ttn #515: Programmatic way to disable question about testing payload functions

joscha, 2017-03-11:
This is a **feature request** for **ttnctl**.

I am running
```console
ttnctl applications pf set converter "./src/ttn/converter.js"
```
from a build script and I always get prompted:
`Do you want to test the payload functions? (Y/n)`
however as it is an automated script, I'd like to disable this prompt: https://github.com/TheThingsNetwork/ttn/blob/ff54fc6b33763f800cf0a86ded8831bf30a88344/ttnctl/cmd/applications_pf_set.go#L160

So I wondered if such a flag can be added or whether I can contribute that and it would be merged.

## Environment
```
INFO Got build information                    Branch=master BuildDate=2017-02-08T11:11:20Z Commit=3c8b184a9a99573b93e89be9d7b2a3d284df376a Version=v2.5.1-dev
```

htdvisser, 2017-03-13:
Sure! Please contribute!

By the way, did you know that you could also use the [Application Manager API](https://www.thethingsnetwork.org/docs/applications/manager/api.html#setapplication) to set these payload functions?

joscha, 2017-03-15:
@htdvisser ah, that is even better - hadn't seen that before - do you know if anyone tried to build a terraform plugin around that?

joscha, 2017-03-15:
I opened #523 to fix this issue, please take a look @htdvisser

closed 2017-03-20 by htdvisser