TheThingsArchive/ttn #519: Pluggable raw payload handling

johanstokking, 2017-03-14:
This is a **feature request** for **the Backend/ttnctl/the Console**.

Currently, we have the payload functions using a JavaScript sandbox.

Alternatively, we need support for other formats. For example, the user would select `TheThingsMessage` (protobuf), `CayenneLPP`, `Custom (JavaScript)`, etc, where the latter is the current implementation.

This should be as dynamic as possible, potentially leveraging Go 1.8's plugin model. This also makes it easier for the community to add payload handlers without building the back-end from source.

htdvisser, 2017-03-14:
- Can't these extra formats be implemented as "packaged" JavaScript functions?
- How is using Go 1.8's plugin model easier than building the backend from source?
- How would these "payload processor plugins" be plugged in, let's say, a community Handler?

johanstokking, 2017-03-14:
Not all formats can be implemented that way, at least in a feasible manner. An example is protobufs.

A plugin model would allow using binaries, whether or not built from source, but at least being able to discover and load formats without (re)building the backend.

In the public community network's Handler, these plugins would be loaded if they're contributed to this repository.

htdvisser, 2017-03-14:
For the record: I'm against using plugins (at least, for now). Plugins are currently only supported on Linux, and would therefore complicate development and restrict us in deployment options.

htdvisser, 2019-02-18:
This will be supported in v3 through the `FORMATTER_GRPC_SERVICE`

closed 2019-02-18 by htdvisser