TheThingsArchive/ttn #520: Rename region to frequencyPlan in gateway status proto

johanstokking, 2017-03-15:
This is a **change request** for **the Backend**.

https://github.com/TheThingsNetwork/ttn/blob/develop/api/gateway/gateway.proto#L82

Should be `frequencyPlan`

htdvisser, 2017-03-20:
Note: This will be a breaking API change for the JSON API (`status.region`) and Go API (`status.Region`). Binary representations of the protos will remain unchanged.

Possibly affected (apart from this repo):

- gateway-connector-bridge ( @htdvisser )
- noc ( @rvolosatovs )
- value-exchange ( @jankramer )
- console ( @romeovs )

jankramer, 2017-03-20:
Not used by VX, so no issues there.

romeovs, 2017-03-20:
Console uses protos so it should be good.

rvolosatovs, 2017-03-20:
Not used in NOC

johanstokking, 2017-03-20:
It will break the `ttn-gateway-connector` when protos are regenerated, but that's a really minor change for users

commit 8884eba6 (2017-03-20): Rename Region to FrequencyPlan

Resolves #520
PR #529 (2017-03-21): Rename Region to FrequencyPlan
Resolves #520
closed 2017-03-21 by htdvisser