TheThingsArchive/ttn #527: Console: Add gateway to the meta information of the downlink message

h0l0gram, 2017-03-19:
This is a **{feature request}** for **{the Console}**.

- Can the meta information of a downlink message, seen in the data section of a device contain the gateway's EUI/id the downlink message was sent to?

I guess that in theory, the gateway received the last uplink message of the device should be the sender of the next downlink message. If more than one gateway received the last message, this becomes even more interesting.

Thanks

h0l0gram, 2017-03-19:
And by the way: Keep up the awesome work with the console! Great improvements compared to staging!
:1st_place_medal:

htdvisser, 2017-03-20:
The event published on MQTT already contains the `gateway_id` field, so this shouldn't be much work, right, @romeovs?

romeovs, 2017-03-20:
done!

htdvisser, 2017-03-20:
[image]

closed 2017-03-20 by htdvisser

h0l0gram, 2017-03-20:
Sorry. Do you mind renaming "gateway_id" to "**gtw_id**". This way, uplink and downlink message will have the **same field name** for the gateway's id.
Beside that it works! Thanks!

htdvisser, 2017-03-20:
That would be a breaking change with no backwards compatibility. Although the event is probably not used by many clients, I don't think we should just introduce breaking changes to our existing APIs just because `gtw_id` is more consistent.

What do you think, @johanstokking?

h0l0gram, 2017-03-20:
Ah, so the chosen different field names are taken directly from mqtt already? Then, never mind.
I thought before people start using the "gateway_id" I ask for the same names. But when the field are different in mqtt already, let's forget it.

johanstokking, 2017-03-20:
Yeah let's stay compatible here for now and let's avoid these inconsistencies.