TheThingsArchive/ttn #533: Devices can have AppEUIs not associated with their application

matthijskooijman, 2017-03-21:
This is a bug report / confusion report for the backend and/or ttnctl.

It seems that you can register a device using an AppEUI that is not associated with the application you're using, without any restriction, which might not be intended and can be confusing. Here's what I did:

 - I registered a new application 'meet-je-stad' on the production backend using ttnctl. It got the 70B3D57EF00040A3 AppEUI assigned. This also selected this application as the default in ttnctl (and, as I later learned, also the AppEUI).
 - Since I was migrating from staging, I used the console webinterface to add my original AppEUI (70B3D57ED00003BA) and remove the new one.
 - So now I only had a single AppEUI assigned to the application:
```
$ ttnctl applications info meet-je-stad
  INFO Found application
AppID:   meet-je-stad
Name:    Meet je stad! dataverzameling
EUIs:
       - 70B3D57ED00003BA
[snip]
```
 - Then I registered a new device:
```
$ ttnctl devices register 50 0000000000000032 ****
  INFO Using Application                        AppEUI=70B3D57EF00040A3 AppID=meet-je-stad
  INFO Discovering Handler...                   Handler=ttn-handler-eu
  INFO Connecting with Handler...               Handler=eu.thethings.network:1904
  INFO Registered device                        AppEUI=70B3D57EF00040A3 AppID=meet-je-stad AppKey=**** DevEUI=0000000000000032 DevID=50
```

As you can see, this registered a device using the original AppEUI, which is no longer associated with the application. I'm not sure if this should be allowed, but it is at least confusing and a warning might be in order.

To solve this, I re-ran `ttnctl applications select` to re-select the application, which also implicitly reselected the AppEUI and made things work.

## Environment

```
$ ttnctl version
  INFO Got build information                    Branch=master BuildDate=2017-03-14T12:01:59Z Commit=a6a96eb34128e21eb459d3396f7992e9fe00567a Version=v2.5.4-dev
  INFO This is an up-to-date ttnctl build
```

htdvisser, 2017-03-21:
It is indeed possible to register devices with any AppEUI when overriding the "selected" AppEUI. However, this should not happen by default. Is it possible that ttnctl was still using an old ttnctl v1 config with your old AppEUI "in use"?

matthijskooijman, 2017-03-21:
I don't think so, since I wiped my `.ttnctl` directory for other reasons. I am still using an old ttnctl version to also talk to staging, but there only the "correct" AppEUI exists, so that cannot have set the incorrect one. To me, it seems like that when selecting an application using ttnctl, it selects both the appid as well as the appeui, and then changing the appeui using the console (not ttnctl) means the appeui selected by ttnctl is still the old one?

htdvisser, 2017-03-21:
`ttnctl` stores the AppID and AppEUI that it uses in `~/.ttnctl/app`. Changing this in the console, won't change this on your harddisk.

matthijskooijman, 2017-03-21:
Obviously, no. I think the main reason for the confusion is that I did not *expect* it to store an AppEUI locally, since I just selected an application. When creating a device in an application, I would expect it to use the application's current AppEUI (moreover, I initially assumed that when an application had multiple AppEUIs, they would be interchangeable, but it turns out that each device is associated with a single AppEUI instead).

In any case, going forward, perhaps a warning could be shown when the currently selected AppEUI is no longer associated with the current application (or perhaps when a device is registered using an AppEUI that is not associated with the current application)? That would have prevented my confusion, without having to change any behaviour.

johanstokking, 2017-05-03:
Yeah it's a bit confusing when using `ttnctl`. In the Console we masked it with the dropdown. In reality, the Handler doesn't care about the AppEUI at all, as it should. Also, the AppEUIs that you configure in your account (stored in the Account Server) only have the purpose to fill the dropdown or select an AppEUI in ttnctl, but they are not unique to applications nor checked by the Handler.

So ttnctl indeed requires you to select one of the app's EUIs, and this EUI is used to register new devices for convenience. I think this is expected behavior.

closed 2017-05-03 by johanstokking